Koenkk/Z-Stack-firmware #10: Offer multiple versions based on high-mesh networks vs single-node

glentakahashi, 2018-09-19:
Was fiddling around with Z-stack 1.2.2 firmwares and was able to compile a version that accepts 44 devices and so far has been stable. Will obviously want to do more testing, but for now my notes below:

Decreasing XDATA stack offset to 0x300 (default) gives a good amount of extra ram and allows approx 20 more devices to be joined. The default in my IAR project was 0x4D0, but this note here http://processors.wiki.ti.com/index.php/Zigbee_Known_Issues_and_Proposed_Fixes#Soft_Reset.2FFreezing.2FStack_Overflow_due_to_BDB_REPORTING_on_CC2530.2FCC2531 makes me believe that the issue arises when BDB_REPORTING is enabled and I don't believe it is for the default firmware. I will report if I hit any issues here in my everyday use.

The other thing I did to increase device limit was to decrease MAX_NEIGHBOR_ENTRIES to 1 because I don't have any routers in my network and the space wasn't being used for anything. I believe I can set this to 0, but in the case that I do get a router I just wanted to have to avoid reflashing everything.

glentakahashi, 2018-09-21:
@Koenkk how would you like to add this to the guide/repo/etc. Do you want me to make a PR that includes my pre-built binary and to update the documentation on building it in the readme as well?

khzd, 2018-09-27:
Hi I m close to 15 devices, interested to use this in case of
@koenkk what is the best process ?

Koenkk, 2018-09-28:
@khzd I would recommend adding a router (CC2531 or CC2530 if you want better range).

glentakahashi, 2019-01-23:
Closing this super old issue since this has been implemented

closed 2019-01-23 by glentakahashi