Koenkk/Z-Stack-firmware #14: Router not pairing devices

paskyecijano, 2018-11-28:
Hi,

I have a CC2531 set as a router but I can not get it to act as a router, that is, it behaves like an END DEVICE,

Do you have to do something special? How do I hook up this router?

Thank you

[image]

paskyecijano, 2018-11-29:
I've managed to get the devices connected to the router.

Now I have a question, if when I have a router with connected devices, and I remove that router, those devices are not reconnected to the coordinator and these devices are stupid

I tested with an INNR bulb, and when I have devices connected to the bulb and remove the bulb, those devices only connect to the coordinator.

ptvoinfo, 2018-11-29:
@paskyecijano Devices need some time to find a new/better path to the coordinator. A device starts to search a new path only when it sends data to a coordinator.

kfetro, 2019-03-11:
I have also the same issue. I cannot pair devices to a router. The router is well paired to the coordinator, and communicaiton works. Howerver, devices cannot be paired to the router, those can however be paired to the coordinator, but if I move them to their location, the linkquality is lower than the linkquality of the router.

closed 2020-08-14 by paskyecijano