Koenkk/Z-Stack-firmware #20: Soft reset of firmware

SargonofAssyria, 2019-01-06:
I hope that there will be a soft-reset of the coordinator firmware in the near future.
Because this often happens after o reboot of my RPi, and when I am not nearby I am not able to press the button. And often I have to pull the CC2531 to get control again.
`2019-1-5 16:33:02 - info: Logging to directory: '/var/log/zigbee'
2019-1-5 16:33:08 - info: Starting zigbee2mqtt version 1.0.1 (commit #9380bbc)
2019-1-5 16:33:08 - info: Starting zigbee-shepherd
2019-1-5 16:33:11 - info: Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2019-1-5 16:34:11 - info: Starting zigbee-shepherd
2019-1-5 16:34:14 - error: Error while starting zigbee-shepherd!
2019-1-5 16:34:14 - error: Press the reset button on the stick (the one closest to the USB) and start again
2019-1-5 16:34:14 - error: Failed to start Error: request timeout
    at CcZnp.<anonymous> (/home/pi/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/home/pi/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)
`

Koenkk, 2019-01-06:
Does it work after those 60 seconds?

SargonofAssyria, 2019-01-06:
Sometimes it works after the 60 sec. But most of the time it does not work, only after I pull the coordinator from the USB and put it back again. And the reset button on the stick mostly does not work either.
Working on a RPi 3B+ with Linux domoticz 4.14.83v7-aufs #1 SMP Wed Nov 28 22:52:43 CET 2018 armv7l GNU/Linux

AnhDuc85, 2019-01-07:
same here,
have to unplug the stick or perform a reboot (NUC), even a restart of the docker container doesn't work

closed 2020-08-18 by github-actions[bot]