Koenkk/Z-Stack-firmware #22: Add possibility to set router into pairingmode without reflash

chaertl1, 2019-01-08:


chaertl1, 2019-01-08:
Is it possible to implement a possibility to switch the router back to pairing mode.
For example at the CC2531 you could use a long button press to activate pairing mode.
On the other devices a GPIO pin connecdtion to ground could be a possibility.

The reason for this is I changed the channel in iobroker config for the coordinator from cannel 11 to channel 26 and back to channel 11 which caused the routers not to connect to the coordinator any more.

If I reflash the routers and pair them again everything works fine again.

Koenkk, 2019-01-09:
@ptvoinfo would this be possible?

ptvoinfo, 2019-01-09:
No, I've disabled buttons on the CC2530 chip because there are different layouts, and the selected pin for a button can be used or connected to ground/+3V. In some cases, a CC2530 board may have only a few pins.

chaertl1, 2019-01-09:
Another hardware independent option would be a powercycle mechanism:
1. After power on/reset increase a counter that is stored in flash
2. Sleep for 5 sec.
3. Check if counter has reached a certain value e.G. 3 then go into pairing mode.
4. Reset the counter to 0 in flash.
5. Execute Regular program.

With this mechanism pressing a reset button/reset pin   3 times within 5sec the cc2530 enters pairing mode.

I know this feature from some ESP8266 programs

ptvoinfo, 2019-01-09:
The main problem here is the pin number. All board with CC2530 have a different layout. For example, I can add this feature on pin P10, but some board do not connect this pin to a header or may connect it to ground.

chaertl1, 2019-01-09:
For my solution you don't need any hardware pin.

ptvoinfo, 2019-01-09:
Hmm. Good idea, but I need some time to add it to the firmware.

chaertl1, 2019-01-09:
Here some code:
```
void main(void){

  int counter = 0;
  readFlash(Address, counter);
  counter++;
  writeFlash(Address, counter);
  
  sleep(5000);
  
  if(counter >=5 ){
    erasePairingData();
  }
  
  counter=0;
  writeFlash(Address, counter);
  
  
  while (TRUE){
  
    //regularCodeHere;  
  
  }
}
```

ptvoinfo, 2019-02-04:
@chaertl1 Implement this in the latest version:
http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/

closed 2020-08-18 by github-actions[bot]