Koenkk/Z-Stack-firmware #23: Coordinator firmware device limits

fensoft, 2019-01-09:
Hello,

What's the limit of the 3 firmwares found in the sourcecode ?
[Here](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/alternatives/max_devices) is written: ~16 devices and ~16 routers for the classic version and 44 devices and 1 router for the alternate version on a single CC2531

If I understand well:
NWK_MAX_DEVICE_LIST = number of total devices
MAX_NEIGHBOR_ENTRIES = number of routers

Questions:
Does a router connected to another router counts as a neighbor (indirect neighbor)?
Does a router counts as a device ?

Koenkk, 2019-01-09:
The normal firmware (which I highly recommend) has support for 15 **direct** nodes. 

I hope this picture explains your question:

[image]

fensoft, 2019-01-09:
Oh, my god. Thanks for a such clear explaination :)

How many total devices does it manage ? Virtualy unlimited ?

Koenkk, 2019-01-10:
@fensoft Hard to say, various users have reported that 50 nodes works without problems, I don't have any reports above that (I expect it to work though).

fensoft, 2019-01-10:
Thanks !

So, in the memory of the coordinator, the list of direct devices are stored in the devices but not the ones behind a router ? So, the router have it's own list of devices ?

Koenkk, 2019-01-10:
@fensoft yes and yes.

leobel96, 2019-01-15:
Sorry for the intrusion but do all the light bulbs act as routers? If yes, can a person add as many light bulbs as he wants ?

fensoft, 2019-01-15:
How does the other devices change their path ? Can we set a device to use a specific router ?

leobel96, 2019-01-15:
In my opinion it would be useful to add the device type near each device in this page https://koenkk.github.io/zigbee2mqtt/information/supported_devices.html.

Koenkk, 2019-01-15:
@fensoft The zigbee protocol takes care of this.

p33k, 2019-02-01:
I connected 21 devices (all xiaomi sensors, and switches, no routers) with my CC2531 stick using iobroker zigbee. 22th device won't connect.
Does 21 make sence in any way?
I bough some osram smart plugs to use them as routers and add more devices, but even when I deleted some sensors the smart plugs don't connect.
Do I have to flash the usb stick again and start all over?

closed 2020-08-18 by github-actions[bot]