Koenkk/Z-Stack-firmware #24: Pi Connection

c3reeman, 2019-01-09:
Hi,

Ive successfully flashed a cc2530+cc2591 im now trying to connect to my Pi running hassio but i'm getting errors in the zigbee2mqtt log.

is my connection correct i'm using a FTDI TTL and 4 wires to GND, VCC , P0_2 , P0_3 thats it am i missing anything ????

I have a CC2531 already flashed and thats working great with the Pi but i wanted swap it for something with more range

Koenkk, 2019-01-10:
@kirovilya do you know how to connect this?

kirovilya, 2019-01-10:
For chips cc2530 usually still need to connect P2_0 to GND. And sometimes P0_4 to GND too.
@c3reeman Did you have chosen correct firmware for cc2530+cc2591 ?

c3reeman, 2019-01-10:
I’ll give Connecting P2_0 to gnd a go thanks,

I used 

https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2530_CC2591

universal, 2019-02-01:
Hmm, as I've got a cc2530+cc2591 as well and was wondering this as well, so after some googling and trial and error this kinda works for me (not yet in hassio, but that seems to be broken right now for me anyways...)

following: https://pinout.xyz/pinout/uart#

i connected:

pi              | cc2530+cc2591
--------------|----------------
pin 9 (gnd)| gnd
pin 8 (txd) | p0_2
pin 10 rxd) | p0_3
ping 17     | vcc

Then i added ```enable_uart=1```to the /boot/config.txt file and then it should be available at ```/dev/ttyS0```, and changing the serial port to that should be fine!

universal, 2019-02-01:
> I’ll give Connecting P2_0 to gnd a go thanks,
> 
> I used
> 
> https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2530_CC2591

just noticed now: since you are connecting it to the pi i think you want to use it as a coordinator, not a router, therefore you need to use the firmware at: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2530_CC2591

universal, 2019-02-02:
Works now for me with hassio also, though i have to manually restart the module, as in replugging the 3.3/vcc connection after hassio has booted, otherwise the zigbee2mqtt service won't start. If there are questions, I'd be happy to help :-)

closed 2020-08-18 by github-actions[bot]