Koenkk/Z-Stack-firmware #25: Router firmware without/with configurable LED

ohrensessel, 2019-01-10:
Currently, the LED of the router is blinking during normal operation. This might get quite annoying depending on where the stick is operating. For the coordinator, zigbee2mqtt offers a handy option to disable the LED.

Would it be possible to disable the LED on the router as well, e.g. by a long press on SW1 or so?

Koenkk, 2019-01-10:
@ptvoinfo is this possible?

ptvoinfo, 2019-01-10:
Currently, the router blinks when it sends the "onOff" state to the coordinator. It is possible to disable this report through the config command. Below is the example, but I don't know how to implement this in the latest version of zigbee2mqtt.

```
function configure_report_interval(dev) {
    'use strict';

    if (!dev) {
        return;
    }

    var info = shepherd.info();
    if (info.joinTimeLeft) {
        setTimeout(configure_report_interval, info.joinTimeLeft * 1000, dev);
        return;
    }

    var sendCB = function (err) {
        if (err) {
            console.log('Unable to configure reporting:', get_device_desc(this.device), 'Err:', err);
        }
    };

    var cId = 'genOnOff';
    var attrId = 'onOff';
    var reportIntervalSec = 60; // 65535 - disables the report at all
    var epId, ep;
    for (epId in dev.endpoints) {
        ep = dev.endpoints[epId];
        if (ep.clusters.hasOwnProperty(cId)) {

            console.log('Sending config report to:', get_device_desc(dev.ieeeAddr));
            ep.report(cId, attrId, reportIntervalSec, reportIntervalSec ,
                sendCB.bind({
                    device: dev.ieeeAddr
                }));
            break;
        }
    }
}
```

ohrensessel, 2019-01-10:
Is this report in any way associated with the link quality and online/offline status reported via MQTT?

I'm referring to the data which is displayed in my homeassistant installation like this:

[image]
[image]

Link quality seems to work well, but the router is always displayed as offline (as you can see in the second image). But this maybe is something for a different issue?

ptvoinfo, 2019-01-10:
@ohrensessel The router reports the state value, but this value is not related to the offline/online state.

d1slact0r, 2019-01-23:
how do I disable the led of the coordinator firmware on my usb stick?

ohrensessel, 2019-01-24:
Via the disable_led config option (see
https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html)

sonntam, 2019-06-19:
@ptvoinfo I tried the following command

```
{
    "devId": "zigbee.0.00124b001936aa9b",
    "ep": "8",
    "cid": "6",
    "cmd": "6",
    "cmdType": "foundation",
    "zclData": {
        "attrId": "0",
        "minRepIntval": "65535",
        "maxRepIntval": "65535",
        "direction": "0"
    }
}
```

but it didn't work. The LED is still blinking. Am I missing something?

ptvoinfo, 2019-06-19:
@sonntam Is the endpoint number correct? Also, it should be "report-config" command.

sonntam, 2019-06-19:
The cmd-id no. 6 corresponds to "configReport" so I think it should be correct. I also tried endpoint no. 0 to no avail.

furiaceka87, 2019-07-27:
Are you able to disable the red blinking led for the router?

sonntam, 2019-07-27:
@furiaceka87 I wasn‘t able to disable the blinking yet. My WAF is really bad right now 😭 
I‘m still hoping for some help.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-08-18 by github-actions[bot]