Koenkk/Z-Stack-firmware #28: cc2530 max devices

lenisko, 2019-01-23:
Hi there!

Is this possible to build & flash `max_devices` fw version to cc2530? same as for cc2531

Greetings!

Koenkk, 2019-01-24:
I would highly recommend adding another router instead of a different firmware.

lenisko, 2019-01-24:
@Koenkk So it was tested and it's not as stable as cc2531 on `max_devices` ? Thank you for opinion.

lenisko, 2019-01-24:
@Koenkk all kind of devices are affected by lack of back paths? (I'm missing knowledge in there)
I had freezing on `master` fw as well, trying change status of 5 bulbs ( back then not connected by group https://github.com/Koenkk/zigbee2mqtt/issues/764#issuecomment-457063073 )

Anyway, I think I will take your advice and add another router. What is making me worry is the fact that I'm expecting about 32 devices (which is out of limit for one coordinator and router)

Koenkk, 2019-01-24:
@lenisko reading that you have 5 bulbs, these will probably also serve as a router. Perhaps https://koenkk.github.io/zigbee2mqtt/information/zigbee_network.html is an interesting read for you.

lenisko, 2019-01-24:
@Koenkk Thank you for pointing that! So stock firmware on `LED1545G12` are good enough to serve as routers in `zigbee2mqtt` based network?

Koenkk, 2019-01-24:
Yes

lenisko, 2019-01-24:
@Koenkk thank you for your help and pointing out my oversight. With this I'm set in coverage area :-)

closed 2019-01-24 by lenisko