Koenkk/Z-Stack-firmware #30: Router firmware is open for new device joins on powercycle.

Adminiuga, 2019-01-31:
Hi,
I'm running CC2531 with router firmware to extend coverage for my Xiaomi devices, but the problem I'm having that whenever CC2531 is power cycled, it opens network for new device joins which pretty much can expose the entire network. 

Is there a setting for the router firmware not to allow joins on restart?

Koenkk, 2019-01-31:
@ptvoinfo is this some setting in the firmware?

Adminiuga, 2019-01-31:
@ptvoinfo I don't think a Zigbee router just re-transmits all packets to coordinator. For a brand new (or factory reset) zigbee device to join the network, it must find a router which "permits" new joins. A Zigbee router can be told to "permit new joins" by sending it a ZDO 'Mgmt_permit_joining_req' command Cluster_ID==0x0036 only after that a new device can join the network and fetch the network key from the trust center. 

So I don't have a direct way to verify it, but indirectly it goes like this:
1. Normally I can't just join a new device to zigbee network
2. I remove an existing device (color bulb) or factory reset it. Upon power cycle, it just changes color indicating that it is not joined to the network. No matter how many times I powercycle it, or how long I wait it won't join the network
3. The bulb joins the network only after I call for "zha.permit" service
4. Alternatively I just power cycle a CC2531 router and bulb joins the network without me calling `zha.permit`. If I remove the bulb from the network it immediately rejoins it, because CC2531 is still open for new joins. In the traces I can see that the reported parent (the router which allowed the new device into the network) is CC2531 router

let me do some more digging to see if I could get a `neighbor table` from an adjacent device

Adminiuga, 2019-01-31:
Unfortunately the "neighbor table' from an adjacent device has "unknown" status for new joins. But I was able to reproduce and confirm issue again:
1. power cycle CC2531 router
2. remove a device from Zigbee network
3. it immediately joins again through CC2531 router

If a send `Mgmt_permit_joining_req` ZDO request with a duration of 10s, then after 10s CC2531 stops accepting new joins. So it behaves correctly in respect to ZDO mgmt_permit_joining behavior, just for whatever reason it defaults to allowing new joins on a power cycle.

Adminiuga, 2019-02-01:
@ptvoinfo any chance you could provide a `.patch` file for CC2531 router firmware? 
Trying to compile it, but without too much luck for now. Is there any reason you went with Z-Stack 1.2.2 instead of 3.0.2?

ptvoinfo, 2019-02-01:
@Adminiuga 
Could you please try the following build?
http://ptvo.info/wp-content/uploads/2018/09/cc2531_1.2.2a.44539_firmware_20190201.zip

Adminiuga, 2019-02-02:
I can't reproduce the issue anymore with the new firmware. Will update my other routers and do more testing, but so far looks good!

ptvoinfo, 2019-02-02:
@Adminiuga Does the router allows you to pair a device after a command from the coordinator?

Adminiuga, 2019-02-04:
@ptvoinfo yes, I send 0x0034 ZDO request specifically to CC2531 and pair my Xiaomi sensors. Using the routers mostly for xiaomi devices, as those are picky about their parent devices.

Adminiuga, 2019-02-04:
I've updated both of my CC2531 routers and so far everything is running good. I consider this fixed and going to close this one.

closed 2019-02-04 by Adminiuga

abmantis, 2019-09-30:
@ptvoinfo @Adminiuga are the firmware files in this repo updated with this fix? They are from 09 2018.