Koenkk/Z-Stack-firmware #31: Bin firmware file

allofmex, 2019-01-31:
Please provide bin file for updates without cc debugger too.
(As mentioned here: [https://github.com/Koenkk/zigbee2mqtt/issues/320](https://github.com/Koenkk/zigbee2mqtt/issues/320))

Carbon459, 2019-02-05:
Yes please, flashing with an Arduino is a hassle. The SerialBootTool is much better.

Koenkk, 2019-02-06:
Done!

closed 2019-02-06 by Koenkk

Anon0ne, 2020-04-22:
For router as well

h4nc, 2020-06-15:
@Koenkk could you please provide the bin files for the [CC2531 router](https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2531) too?

Koenkk, 2020-06-15:
@ptvoinfo would it be possible to provide this?

ptvoinfo, 2020-06-15:
@Anon0ne It is here: https://ptvo.info/cc2531-based-router-firmware-136/

h4nc, 2020-06-25:
@Koenkk @ptvoinfo I think I noticed that the bin files wouldn't make sense for the router. I tried flashing a router to a coordinator (with SerialBootTool) and I experienced that the USB port of the router doesn't get recognized on multiple machines. 

So my guess would be that the router firmware doesn't include the needed USB port connection to be able to use the SerialBootTool. Is that true?

ptvoinfo, 2020-06-26:
@h4nc The router firmware does not include USB features because it is useless for a router. Also, SerialBootTool cannot upload new firmware through USB.

h4nc, 2020-06-26:
@ptvoinfo thanks for your answer. 

> The router firmware does not include USB features because it is useless for a router. 

Ok, so it absolutely makes sense that there are no bin files available for the router firmware, because updating a router with the SerialBootTool wouldn't work. I only see one case that could work. Flashing a coordinator to a router. But I understand that it doesn't make much sense to provide files for this case, because flashing would be one way (meaning if it would work you couldn't flash it again with the SerialBootTool).

> Also, SerialBootTool cannot upload new firmware through USB.
You mean that the SerialBootTool is only able to update devices and not to do the very first flashing process, right?

ptvoinfo, 2020-06-26:
@h4c SerialBootTool: Yes, the previous flashed firmware should support it. It works through UART pin P02&P03 (not USB).

h4nc, 2020-06-26:
> It works through UART pin P02&P03 (not USB).

With that you mean the first/previous flashing process needs to be done through UART (for example with the CCDebugger), right?

Afterwards you can use the SerialBootTool which works with USB (of course only if the firmware supports it -> coordinator)

h4nc, 2020-06-26:
> It works through UART pin P02&P03 (not USB).

@ptvoinfo With that you mean the first/previous flashing process needs to be done through UART (for example with the CCDebugger), right?

Afterwards you can use the SerialBootTool which works with USB (of course only if the firmware supports it -> coordinator)

ptvoinfo, 2020-06-26:
@h4nc Generally, the first time you should flash using CCDebugger (or alternative flashing method). CCDebugger does not use UART pins. Afterward, you can use the SerialBootTool tool.

h4nc, 2020-06-26:
@ptvoinfo thanks for clearing that up and sorry that I got your last message wrong. So the SerialBootTool works with UART. I was confused by 

> (not USB)

And I just reread [this](https://github.com/Koenkk/zigbee2mqtt/issues/320#issue-355660026):

> ZNP firmware for cc253* has a built-in bootloader that can be used to updated firmware via USB / UART into the cc2531 stick