Koenkk/Z-Stack-firmware #33: git apply patch

yaniv1111, 2019-02-04:
Hi,

Im trying to apply "fireware.patch" file and got this errors

MINGW64 /c/Texas Instruments/Z-Stack Home 1.2.2a.44539                             $ ls                                                                                           _iss/                                                                                         Accessories/                                                                                  Components/                                                                                   Documents/                                                                                    EULA.pdf                                                                                      firmware.patch                                                                               'Getting Started Guide - CC2530.pdf'                                                          'Getting Started Guide - CC2538.pdf'                                                           Projects/                                                                                     Tools/                                                                                       'Z-Stack Core Release Notes.txt'                                                              'Z-Stack Home Release Notes.txt'                                                              'Z-Stack Home Software Development Kit Resource Guide.html'                                    Z-Stack_Home_1.2.2_Manifest.pdf                                                                                                                                                            YK@WIN7-PC MINGW64 /c/Texas Instruments/Z-Stack Home 1.2.2a.44539                             $ git apply firmware.patch                                                                    firmware.patch:41: trailing whitespace.                                                       #define CODE_REVISION_NUMBER 20190109                                                         firmware.patch:73: trailing whitespace.                                                                                                                                                     firmware.patch:79: trailing whitespace.                                                                                                                                                     firmware.patch:89: trailing whitespace.                                                       // Shared accross all firmwares                                                               firmware.patch:90: trailing whitespace.                                                       #define ASSERT_RESET                                                                          error: patch failed: Projects/zstack/ZNP/Source/znp.cfg:101                                   error: Projects/zstack/ZNP/Source/znp.cfg: patch does not apply
--

thanks
Yaniv

Koenkk, 2019-02-04:
Is this a clean Z-Stack install?

yaniv1111, 2019-02-04:
yes on windows 10

Koenkk, 2019-02-06:
Can you check if this has been fixed?

yaniv1111, 2019-02-06:
yes. this has been fixed by:
dos4unix fireware.patch
and trimming all leading spaces using notepad++

yaniv1111, 2019-02-06:
I'm tring to set FIRMWARE_CC2531_DEFAULT or FIRMWARE_CC2531_MAX_DEVICES but I dont see it in "defined symbols"

[image]

Koenkk, 2019-02-07:
You need to remove all symbols and only add FIRMWARE_CC2531_DEFAULT

yaniv1111, 2019-02-07:
I tried to rebuild but got this errors.

[output.built_all.txt](https://github.com/Koenkk/Z-Stack-firmware/files/2842252/output.built_all.txt)

closed 2020-08-18 by github-actions[bot]