Koenkk/Z-Stack-firmware #34: Coordinator initialize had an error: request timeout

jiki974, 2019-02-07:
Hi, I have follow your instructions with the patch and IAR.
It compile, I can upload hex file but when I try zigbee shepherd I get this error : 

  serialport/bindings loading WindowsBinding +0ms
  zigbee-shepherd starting +0ms
  zigbee-shepherd:init zigbee-shepherd booting... +1ms
  serialport/stream opening path: COM6 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +7ms
  serialport/stream opened path: COM6 +85ms
  cc-znp The serialport COM6 is opened. +99ms
  serialport/stream _read reading +2ms
  serialport/binding-abstract read +92ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +5ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +2ms
  serialport/stream _write 8 bytes of data +20ms
  serialport/binding-abstract write 8 bytes +22ms
  serialport/stream binding.write write finished +9ms
  cc-znp:SRSP <-- SYS:osalNvRead, TIMEOUT, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +4s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +6ms
  zigbee-shepherd:init Coordinator initialize had an error: request timeout +2ms

With the hex that you provide it works well.
Any Idea ??
My IAR version 10.30.1

Koenkk, 2019-02-07:
Did you exactly follow the intructions or did you made some modifications?

jiki974, 2019-02-07:
Yes 

1- I have download the Z-Stack Home 1.2.2a.44539
2- I have copy/paste the patch file in the directory
3- Git apply firmware.patch & I can see the modification on files 
4- Open znp enw in IAR click Yes twice to convert projects
5 - Set target to cc2530-prodhex
6- put only "FIRMWARE_CC2530_DEFAULT" in Defined symbols
7- Change Override default  to CC2530.hex 
8 - Rebuild all
9 - Flash my CC2530 with this hex file

Koenkk, 2019-02-07:
What do you mean with *7- Change Override default to CC2530.hex*?

jiki974, 2019-02-07:
In option -> linker -> output I have to set  override default with a .hex otherwise it generate a .a51 file   ! 
[image]

Koenkk, 2019-02-07:
That shouldn't be needed, probably the reason why it fails, setting target to cc2530-prodhex should produce a .hex file.

closed 2019-02-07 by jiki974