Koenkk/Z-Stack-firmware #36: Why custom firmware?

wsowa, 2019-02-16:
Hi,

Can you explain why do we need a custom firmware? I'm just curious as the default ZNP provided by TI should be enough... Looking at the firmware patch it looks more like a fine tuning of the Z-Stack rather than any significant change. Would be great if you could add to the readme a short paragraph explaining why the custom firmware is required. Thanks.

Koenkk, 2019-02-16:
These are indeed some small changes, e.g. increased memory and some fine tuning. What do you expect to be explained here?

wsowa, 2019-02-18:
Basically what you just replied. Something that will help people to understand whether that have to flash your firmware or is it ok to stay with standard one. What are benefits of one over the other, etc. 
It’s just not clear until you look at the patch and then it still doesn’t tell a lot of the story to someone not familiar with Z-Stack code. 

Thanks for explaining anyway.

Koenkk, 2019-02-19:
What firmware do you mean with *the standard one*, AFAIK there is no standard firmware.

If by standard one you mean the one that is already flashed on the CC2531, this firmware does not contain coordinator functionality.

wsowa, 2019-02-19:
I mean a binary CC2531ZNP-Pro-Secure_Standard.hex file distributed with z-stack 1.2.2a
It is a ZNP so it does support Coordinator, Router and End device modes. This project use it for example https://github.com/zsmartsystems/com.zsmartsystems.zigbee

Koenkk, 2019-02-20:
Added everything here: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/README.md

closed 2019-02-20 by Koenkk