Koenkk/Z-Stack-firmware #37: Add Sniffing

iwl, 2019-02-16:
It would be great if the firmware could be used for sniffing to.

Tune a channel, switch on receiving, no filtering, forward all packets to serial.

May be this is possible already with the SYS/MAC Layer, manual hacking registers/settings with SYS, 
and forwarding of packets from hardware to serial already in MAC.

I've tried to find a simple demo to get mac-receiving running somehow and tweak it to just forward all, 
unfortunately I only find complicated stuff and putting it all together myself will probably take days because it will not work for several reasons I will not now or make right.

Koenkk, 2019-02-16:
Why not just use the existing ti firmware for this? http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

iwl, 2019-02-16:
Having it all together and be more flexible is preferable.

As I said this is may be already possible sending the right tweaks via serial.

Are there tools/scripts I can convert that hex to sbl-format, so I can transfer it with bootloader?

Does the ti sniff firmware has that bootloader too?

iwl, 2019-02-16:
I've also couldn't find what the green LED, or the red means, I can somehow control with the 2 buttons, searched trough several Pdfs.

Koenkk, 2019-02-17:
- The .sbl firmwares are also available in the zip files in master branch.
- I dont know if the packet sniffer also has the sbl, but you can check this easily I guess by opening the ti sbl tool.
- Green means running, red means that the coordinator is not ready (e.g. initializing)

About i tegrating the sniffer

iwl, 2019-02-17:
With the leds behavior is as following with my Koenkk Stick:

When I connect it, green led is on and Z-Tool don't find the device.
After ~ 1 minute green led goes off and then Z-Tool can connect to the device.
I can press the button beside the green led and it goes off immediately and Z-Tool can connect.
If I press other button when green led is on red led goes on and Z-Tool also can't connect.

According the sniffing you already told not being an expert of CC253x / API.
I'm hoping finding somebody here who is.
Tune a channel, switching on receiver, have no filter and just forward every packet the hardware receives 
to serial can't be that difficult when familiar with device and api.
And may be that can be done already changing some registers via SYS calls and no change to firmware at all.

Unfortunately it is quit steep to get into the CC2531, there is no free dev environment, device has no serial boot loader in hardware, you can't find small examples working and change to your needs.

closed 2020-08-18 by github-actions[bot]