Koenkk/Z-Stack-firmware #38: SBL Version of router firmware necessary

tripplet, 2019-02-17:
I would like to use a spare CC2531 stick as a router, which was flashed with the SBL version of the coordinator firmware (by a friend), can I use the current router firmware with the SBL tool.

Will this remove the serial bootloader?

Koenkk, 2019-02-17:
@ptvoinfo could you provide the sbl firmware?

tripplet, 2019-02-22:
@ptvoinfo Any progress expected otherwise I'll try the alternative method to flash with an Arduino.
Just wanna know and avoid the work if possible but if you currently don't have the time its ok too.

ptvoinfo, 2019-02-24:
@tripplet I don't have the SBL version of the router firmware yet.

WildRat, 2020-07-06:
@ptvoinfo, could you, please share  SBL version of the router firmware?

closed 2020-08-18 by github-actions[bot]

Jamjon, 2021-01-13:
would also be interested to change a coordinator to a router with zigpy-znp 

Is there any chance this router bin file will be available soon?