Koenkk/Z-Stack-firmware #41: Safe to buy preflashed?

imbaimbaimba, 2019-02-20:
Hi guys,
I been trying for about 3 months now to get a hold of the parts needed to get started on Zigbee2MQTT, but they never seem to show up from China, and I just end with refunds. I would really like to get going, so I found that you can buy preflashed sticks too. 
Would you say it is safe to buy one like this?
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.de%2Fulk%2Fitm%2F372536176132

I assume I will receive a debugger etc at some point, so I can update when that happens.

iwl, 2019-02-20:
I get >90% of things I ordered at ebay from China, it takes 3-5 Weeks to Germany, surprisingly fast some times. If I order from a ebay seller in Germany it usually take 2 days at 3-4x the price.
Actual ZStacks have a bootloader, so you can update without a programmer too.
CC2531 with Z-Stack are little above 15€ from a German seller, bought one for 19€ with Z-Stack and bootloader, recognized by Z-Tool, haven't second yet to test radio transmission, 30 days money back.

Ingramz, 2019-02-20:
It is probably safe to buy, considering that there is not much malicious you can do to the device. Of course some caution needs to be exercised with any IoT device you bring home, but if you want to be absolutely certain that it is the firmware you expect it to be, you can always flash it yourself.

I find it incredibly amusing that within a couple of months, these sticks have been selling at a rate of 2-3 per day at >4x the original price.

iwl, 2019-02-20:
Also have a look at AliExpress:

https://www.aliexpress.com/premium/CC2531.html

iwl, 2019-02-22:
If ordered following (CC2531 Sniffer) 8.Feb and got it today 22.Feb, that's 2 weeks from China:

https://www.ebay.de/itm/232543056621

Sniffer FW is installed and just works with TI PacketSniffer (Not 2 !! , the older one from 2014)

henriz, 2019-03-19:
Just order using Aliexpress. If you're in a hurry, take the hit on the postage, and you'll have it within a week. If you're not in a hurry, use standard shipping, but expect it to take a month at least to arrive at your doorstep, depending on the customs in your country. 

I have ordered several this way, and the programmer and programmer cable, and they always got here. Actually the sellers usually ship about the same day, it's the customs at the destination country (in my case the Netherlands) who deliberatelly keep these packages very long (market protection).

Also note that you need to order 2 things to programm it, the programmer AND an extra cable, as the pitch of the connector on the 'standard' programmer is different to the pitch of the PCB's, so you need an adaptor board with a smaller pitch ribbon cable.

millerce, 2019-03-22:
I have two that I can part with that have been flashed to latest. Beware of aliexpress though. One of the ones I bought about fried my $5k computer and they wanted pictures as proof- like i'm going to send them a picture of my dead computer! Needless to say, I got ripped off.

closed 2020-08-18 by github-actions[bot]