Koenkk/Z-Stack-firmware #42: Will it work ?

dgomes, 2019-02-20:
https://www.aliexpress.com/item/Zigbee-CC2531-4dBm-Wireless-Transceiver-E18-2G4U04B-USB-Connector-IO-Port-IoT-PCB-Antenna-2-4GHz/32973371661.html?spm=2114.search0104.3.229.37cf5b58RUjyGL&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10068_319_10059_10884_317_10887_10696_321_322_453_10084_454_10083_10103_10618_10304_10307_10820_10301_10821_537_536_10902,searchweb201603_45,ppcSwitch_0&algo_expid=1eb12bb0-6133-4c11-8e1b-11dc6e05dda7-34&algo_pvid=1eb12bb0-6133-4c11-8e1b-11dc6e05dda7

Koenkk, 2019-02-20:
Probably yes, only connecting the CC debugger is more complicated here (and requires soldering).

iwl, 2019-02-20:
Is there a flashed firmware to be expected (I read nothing) or not and no function without flashing something with CC-Debugger. In later case why not buying one with usual 1.27 10 Pin Connector?

closed 2020-08-18 by github-actions[bot]