Koenkk/Z-Stack-firmware #45: Alternative firmware cc2530

JLFN, 2019-02-23:
@Koenkk  it is possible to make an alternative cc2530 firmware that works with rts flowcontroll i have a Uart with FDTi chip that suppose to work good with rts flow control. It also have Cts if it can be used .

iwl, 2019-02-23:
To what the cc2530 is connected? 
Isn't FTDI USB-Serial? Why not using CC2531 with it's own USB?
And even with FTDI USB-Serial a USB-Host shouldn't have any problems with 250kBit max from ZigBee.

JLFN, 2019-02-23:
i have cc2531 already i want to use cc2530 because of the antenna

iwl, 2019-02-23:
Isn't communication by packets, send one, get an answer and so on, that's a handshake already.
And packets are around 127 Byte + some control bytes.
So I guess there is no need for additional handshake?
I'm not sure if CC2530 sends packets he receives on it's own, but usually host has more memory and 250kbps isn't that fast, so there should be no need to stop CC2530.

closed 2020-08-18 by github-actions[bot]