Koenkk/Z-Stack-firmware #46: cc2531 crashes with the coordinator dev firmware

smarti2019, 2019-02-23:
currently there is a reproducible problem with different cc2531 chips with the dev firmware. After a few hours, the chip crashes and no communication takes place. First unplug the chip and plug it back in to fix the problem.

Koenkk, 2019-02-23:
what firmware version are you running?

smarti2019, 2019-02-23:
i have the dev firmware 2019-02-19 for the cc2531 coordinator flashed

iwl, 2019-02-23:
Koenkk> what firmware version are you running?

I asked you to add Version numbers to your Commits/Branches, you only told the Version it is derived from, to find in some subdir.
I still not get that to be honest. 
Aren't there changes of your changes too, so some Version would be good.
I think smarti2019 has same problem so can only tell a date.

Koenkk, 2019-02-23:
@iwl the coordinator firmware version is logged on startup of zigbee2mqtt. This number is changed for every new firmware version.

@smarti2019 firmware `20190219` doesn't exist, can you post your zigbee2mqtt startup log?

smarti2019, 2019-02-23:
2019-02-23 16:02:15.818  - info: zigbee.0 zigbee-shepherd ready. version: 2.6.3 rev 20190218
 
that is the information from log from iobroker zigbee adapter

Koenkk, 2019-02-23:
@smarti2019 can you try with `20190222` (https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin)

smarti2019, 2019-02-23:
ok, I test with the mentioned firmware and will report

smarti2019, 2019-02-24:
after 24h no problems, no disconnects, all OK!

smarti2019, 2019-02-25:
The problem does not seem to be solved yet. This morning the chip has crashed again

Koenkk, 2019-02-25:
When does this happen? After a burst of commands? Are you sure it doesn't happen with the `20190109` firmware?

smarti2019, 2019-02-25:
no, the problem is with the current dev firmware from 22/23 / February

mullischlumpf, 2019-02-25:
Hi i have the same problems and use the Firmware CC2531ZNP-Prod_20190109 Version on my stick.

smarti2019, 2019-02-25:
I flashed the 20190109 Version for the CC2531 coordinator.
So I paired a few devices and after about 40 minutes hangs on the cc2531.
where can i download the last functions stable version of cc2531 coordinator firmware form 2018?

Koenkk, 2019-02-26:
@smarti2019 https://github.com/Koenkk/Z-Stack-firmware/tree/f4f0e66902eb143c59f1c5208d4ef61094ba344e/coordinator/CC2531/bin 

But I highly doubt this wil solve your problem, probably some other process on your PC tries to access to usb port, because of this communcation fails. You should find out which process does this.

brtwrs, 2019-02-27:
Im having exactly the same problems. Using this project for months now and havent seen this problem even once. Yesterday I flashed the 23-2 version and now I cant get it to work. Keeps giving time-outs after 1-2 minutes or crashes. Even flashing back older versions like the 19-01 wont help. Tried both the master and dev zigbee2mqtt versions too. I looks like the stick crashes or stops responding or something like that.

Koenkk, 2019-02-27:
@brtwrs as downgrading doesn't solve the problem, probably something else is going on (which has nothing to do with the firmware). Can you post your zigbee2mqtt log and the output of `ls /dev`?

brtwrs, 2019-02-27:
Here is my log:

`2019-2-27 11:09:23 - info: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0001656eb5/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/bemotion","json_attributes_topic":"zigbee2mqtt/bemotion","name":"bemotion_occupancy","unique_id":"0x00158d0001656eb5_occupancy_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0001656eb5","name":"bemotion","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-27 11:09:23 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0001656eb5/illuminance/config', payload '{"unit_of_measurement":"lx","device_class":"illuminance","value_template":"{{ value_json.illuminance }}","state_topic":"zigbee2mqtt/bemotion","json_attributes_topic":"zigbee2mqtt/bemotion","name":"bemotion_illuminance","unique_id":"0x00158d0001656eb5_illuminance_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0001656eb5","name":"bemotion","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-27 11:09:23 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0001656eb5/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/bemotion","json_attributes_topic":"zigbee2mqtt/bemotion","name":"bemotion_battery","unique_id":"0x00158d0001656eb5_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0001656eb5","name":"bemotion","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-27 11:09:23 - info: MQTT publish: topic 'homeassistant/light/0x90fd9ffffe16ad09/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/bespot2/set","state_topic":"zigbee2mqtt/bespot2","name":"bespot2_light","unique_id":"0x90fd9ffffe16ad09_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x90fd9ffffe16ad09","name":"bespot2","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI LED bulb GU10 400 lumen, dimmable (LED1650R5)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-27 11:09:23 - info: MQTT publish: topic 'homeassistant/light/0x000b57fffe9a8c12/light/config', payload '{"brightness":true,"color_temp":true,"schema":"json","command_topic":"zigbee2mqtt/alight/set","state_topic":"zigbee2mqtt/alight","name":"alight_light","unique_id":"0x000b57fffe9a8c12_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000b57fffe9a8c12","name":"alight","sw_version":"Zigbee2mqtt 1.1.1","model":"TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (LED1545G12)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-27 11:09:23 - info: MQTT publish: topic 'homeassistant/binary_sensor/0x00124b0009eb0446/router/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.state }}","device_class":"connectivity","state_topic":"zigbee2mqtt/0x00124b0009eb0446","json_attributes_topic":"zigbee2mqtt/0x00124b0009eb0446","name":"0x00124b0009eb0446_router","unique_id":"0x00124b0009eb0446_router_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00124b0009eb0446","name":"0x00124b0009eb0446","sw_version":"Zigbee2mqtt 1.1.1","model":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)","manufacturer":"Custom devices (DiY)"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-27 11:09:23 - info: MQTT publish: topic 'homeassistant/sensor/0x00124b0009eb0446/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00124b0009eb0446","json_attributes_topic":"zigbee2mqtt/0x00124b0009eb0446","name":"0x00124b0009eb0446_linkquality","unique_id":"0x00124b0009eb0446_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00124b0009eb0446","name":"0x00124b0009eb0446","sw_version":"Zigbee2mqtt 1.1.1","model":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)","manufacturer":"Custom devices (DiY)"},"availability_topic":"zigbee2mqtt/bridge/state"}'
2019-2-27 11:09:23 - info: Successfully applied coordinator group for IKEA TRADFRI remote control (0x000b57fffeb0c369)
2019-2-27 11:09:23 - info: Successfully applied coordinator group for IKEA TRADFRI remote control (0x000b57fffeb2b754)
2019-2-27 11:09:23 - debug: Mounted the cieApp (epId 11)
2019-2-27 11:09:27 - debug: Received MQTT message on 'zigbee2mqtt/kled/set' with data '{"state": "ON", "brightness": 178}'
2019-2-27 11:09:27 - debug: Skipping 'state' because of 'brightness'
2019-2-27 11:09:27 - info: Zigbee publish to device '0x000b57fffea15468', genLevelCtrl - moveToLevelWithOnOff - {"level":178,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-2-27 11:09:27 - error: Zigbee publish to device '0x000b57fffea15468', genLevelCtrl - moveToLevelWithOnOff - {"level":178,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)j [... cut ...]
2019-2-27 11:16:34 - debug: Check online kiswitch 0x00158d0001660bfe
2019-2-27 11:16:40 - debug: Failed to ping 0x00158d0001660bfe
2019-2-27 11:16:40 - debug: Check online hswitch 0x00158d0001660bf9
2019-2-27 11:16:46 - debug: Failed to ping 0x00158d0001660bf9
2019-2-27 11:16:46 - debug: Check online 0x00124b0009eb0446 0x00124b0009eb0446
`
At the end you can see there are no more messages en the pings are failing.

And here the output of ls /dev/ttyACM0 -al

`crw-rw-rw- 1 root dialout 166, 0 Feb 27 12:07 /dev/ttyACM0
`

brtwrs, 2019-02-27:
I did some more testing and the same thing is happening on my server, desktop and mediaplayer. (all docker)

brtwrs, 2019-02-28:
update: I turned off (cut the power) some of by ikea bulbs (about 10) and now the network is stable for over an hour, that's way more then the 1-2 minutes I got the last few days. Maybe it has to do with the size of my network.. 

ps sorry @smarti2019 for hijacking your thread ^^

smarti2019, 2019-02-28:
I have also reduced the network as a trial and now extend it succesively. Currently it is stable for 24 hours
[image]

nob0dy80, 2019-03-01:
i'm so sick of this .. for the time spend, i could buy 20 hue bridges.
setup my network with cc2531 with 20190109 firmware.
first time it crashed within 10 hours, the seconds try lasted 26 hours.
light turns off on cc2531 and even after rebooting , the cc2531 led turns off after 2-3 minutes.
reflashed the stick to make it work again. 

my 3. try was a cc2530 over ttl adapter. funny thing: with 20190109 this one is dead after 24 hours too! But this one is working again after a reboot.

one week lost life time till now :-/

brtwrs, 2019-03-01:
How big is your network? I had around 60 devices, with +-30 being able to route. I guess this is just to much. I have now split my netwerk in two, downstairs and upstairs. Works stable since.

nob0dy80, 2019-03-01:
my network was about 30 devices .. ..~20 end, 10 router.
will give this weekend the 20190227 a last try on the cc2531 and i try to connect the cc2530 directly via uart ..
if this two tests fail, i'm done with this and will go back to the good working vendor gateways. :-|

brtwrs, 2019-03-01:
I have two cc2531 sticks with the latest dev version (02-23). Stable as in no more crashed since I set it up yesterday. This is way better then the 1-2 minutes I got before

nob0dy80, 2019-03-01:
well call it stable when its running a week with 30+ devices.

brtwrs, 2019-03-01:
Ill let you know in a week;) In the mean while you could do some debugging and programming to make it more stable.

nob0dy80, 2019-03-01:
ok.. dev firmware 190223 isnt working good for me.
you cant add more then 1 group. if i add a second, i get:
Failed to add group 2 to coordinator. Error: request unsuccess: 178

Koenkk, 2019-03-01:
@nob0dy80 why do you want to add the coordinator to groups?

nob0dy80, 2019-03-01:
i dont want to add the coordinator to a group, i want to add a second group to the coordinator. this happens automatically when you choose "Add group" in Iobroker.
Need this cause i want to turn of more than one light with a switch.

nob0dy80, 2019-03-01:
ok seems like this is the first time i have to give up... 20190223 is working more worse for me than 20190109 stable one.
20190223: crash after 1 hour: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)

There must be something that crashes the coordinator firmware everytime :-(

Koenkk, 2019-03-01:
Why does it restart zigbee-shepherd? @kirovilya do you know why?

nob0dy80, 2019-03-01:
@Koenkk 
since its clear to me now i cant use 02-23 cause of missing groups which are very needed to me for handling brightness in a group, i have to go back to the last version without this change.
can i download 20190218 anywhere?

Koenkk, 2019-03-01:
@nob0dy80 just look at the git history, you will find all previous version there.

nob0dy80, 2019-03-01:
@Koenkk i'm not that pro with github. cant find a history. where can i access it ?

nob0dy80, 2019-03-01:
zigbee.0 | 2019-03-01 17:05:49.763 | info | Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
-- | -- | -- | --
host.iobroker | 2019-03-01 17:05:47.182 | info | instance system.adapter.zigbee.0 terminated with code 0 (OK)
zigbee.0 | 2019-03-01 17:05:46.339 | info | Reset coordinator
zigbee.0 | 2019-03-01 17:05:46.339 | info | Queue is: true
zigbee.0 | 2019-03-01 17:05:46.339 | info | Start on port: /dev/ttyACM0 with panID 6754 channel 11
zigbee.0 | 2019-03-01 17:05:46.335 | info | starting. Version 0.9.2 in /opt/iobroker/node_modules/iobroker.zigbee, node: v8.15.0
host.iobroker | 2019-03-01 17:05:44.157 | info | instance system.adapter.zigbee.0 started with pid 1787
host.iobrokerslave | 2019-03-01 17:05:41.616 | info | object change system.adapter.zigbee.0
host.iobroker | 2019-03-01 17:05:41.633 | info | stopInstance system.adapter.zigbee.0 killing pid 1427
host.iobroker | 2019-03-01 17:05:41.632 | info | stopInstance system.adapter.zigbee.0
host.iobroker | 2019-03-01 17:05:41.631 | info | object change system.adapter.zigbee.0
zigbee.0 | 2019-03-01 17:05:17.075 | error | Zigbee publish to '0x000b57fffebb03eb', genOnOff - on - {} - 1 failed with error Error: Timed out after 30000 ms
zigbee.0 | 2019-03-01 17:05:15.406 | error | No state available for 'AB3257001NJ' with key 'available'
zigbee.0 | 2019-03-01 17:05:07.651 | error | No state available for 'AB3257001NJ' with key 'available'
zigbee.0 | 2019-03-01 17:04:47.069 | error | Zigbee publish to '0x90fd9ffffe1a77dd', genLevelCtrl - moveToLevelWithOnOff - {"level":false,"transtime":0} - 1 failed with error Error: Timed out after 30000 ms
zigbee.0 | 2019-03-01 17:04:17.061 | error | Zigbee publish to '0x000b57fffebb03eb', genLevelCtrl - moveToLevelWithOnOff - {"level":false,"transtime":0} - 1 failed with error Error: Timed out after 30000 ms
zigbee.0 | 2019-03-01 17:03:47.055 | error | Zigbee publish to '0x000b57fffe88fe66', genLevelCtrl - moveToLevelWithOnOff - {"level":false,"transtime":0} - 1 failed with error Error: request timeout
zigbee.0 | 2019-03-01 17:03:44.044 | error | Zigbee publish to '0x000b57fffe846613', genLevelCtrl - moveToLevelWithOnOff - {"level":false,"transtime":0} - 1 failed with error Error: request timeout
zigbee.0 | 2019-03-01 17:03:41.036 | error | Zigbee publish to '0x90fd9ffffe1a77dd', genLevelCtrl - moveToLevelWithOnOff - {"level":false,"transtime":0} - 1 failed with error Error: request timeout
zigbee.0 | 2019-03-01 17:03:38.030 | error | Zigbee publish to '0x000b57fffebb03eb', genLevelCtrl - moveToLevelWithOnOff - {"level":false,"transtime":0} - 1 failed with error Error: request timeout
zigbee.0 | 2019-03-01 17:03:35.020 | error | Zigbee publish to '0x000b57fffe88fe66', genLevelCtrl - moveToLevelWithOnOff - {"level":false,"transtime":0} - 1 failed with error Error: request timeout
zigbee.0 | 2019-03-01 17:03:32.018 | error | Cannot get the Node Descriptor of the Device: 0x000b57fffebb217f (Error: request timeout)
zigbee.0 | 2019-03-01 17:03:29.011 | error | Zigbee publish to '0x000b57fffe846613', genLevelCtrl - moveToLevelWithOnOff - {"level":false,"transtime":0} - 1 failed with error Error: request timeout
zigbee.0 | 2019-03-01 17:03:22.991 | error | Cannot get the Node Descriptor of the Device: 0x000b57fffeb72b00 (Error: request timeout)
zigbee.0 | 2019-03-01 17:03:19.988 | error | Join failure.. Error: request timeout
zigbee.0 | 2019-03-01 17:03:16.355 | info | Zigbee: allowing new devices to join.
zigbee.0 | 2019-03-01 17:03:13.980 | error | Join failure.. Error: request timeout
zigbee.0 | 2019-03-01 17:03:10.979 | error | Cannot get the Node Descriptor of the Device: 0x000b57fffe88fb0e (Error: request timeout)
zigbee.0 | 2019-03-01 17:03:10.540 | info | Zigbee: allowing new devices to join.
zigbee.0 | 2019-03-01 17:03:04.957 | error | Cannot get the Node Descriptor of the Device: 0x000b57fffebb217f (Error: Timed out after 10000 ms)
zigbee.0 | 2019-03-01 17:02:44.937 | info | new dev 0x000b57fffe8af279 40024 TRADFRI bulb GU10 W 400lm

in 3 minutes from a working system to a car crash

smarti2019, 2019-03-03:
have a fresh iobroker installation with zigbee adapter 0.9.2 and dev coordinator firmware from 23 february for 4 days without errors. Now after 4 days it suddenly comes to problems.
see abstract log file:

[67.820326] usb 1-1.2.4: USB disconnect, device number 5
[68.211935] usb 1-1.2.4: new full-speed USB device number 6 using dwc_otg
[83.551898] usb 1-1.2.4: device descriptor read / 64, error -110
[98.991904] usb 1-1.2.4: device descriptor read / 64, error -110
[99.311944] usb 1-1.2.4: new full-speed USB device number 7 using dwc_otg
[114.671983] usb 1-1.2.4: device descriptor read / 64, error -110
[130.111954] usb 1-1.2.4: device descriptor read / 64, error -110
[130.232066] usb 1-1.2-port4: attempt power cycle
[130.991941] usb 1-1.2.4: new full-speed USB device number 8 using dwc_otg
[141.570209] usb 1-1.2.4: device not accepting address 8, error -110
[141.780112] usb 1-1.2.4: new full-speed USB device number 9 using dwc_otg
[152.288413] usb 1-1.2.4: device not accepting address 9, error -110
[152.288553] usb 1-1.2-port4: unable to enumerate USB device
[364.069457] usb 1-1.2.4: new full-speed USB device number 10 using dwc_otg
[364.317222] usb 1-1.2.4: New USB device found, idVendor = 0451, idProduct = 16a8, bcdDevice = 0.09
[364.317235] usb 1-1.2.4: New USB device strings: Mfr = 1, Product = 2, SerialNumber = 3
[364.317245] usb 1-1.2.4: Product: TI CC2531 USB CDC
[364.317254] usb 1-1.2.4: Manufacturer: Texas Instruments
[364.317263] usb 1-1.2.4: SerialNumber: __0X00124B001936B494
[364.319809] cdc_acm 1-1.2.4: 1.0: ttyACM0: USB ACM device
[617.743801] cdc_acm 1-1.2.4: 1.0: failed to set dtr / rts

There is something really wrong. The stick is ok, I exchanged last week.

nob0dy80, 2019-03-04:
@smarti2019 
This one looks very clear to me. Not a stick problem, its the host.
[130.232066] usb 1-1.2-port4: attempt power cycle

if you didnt plug it out, the stick lost its usb connection.
Seen this only once by an raspbian user...
you aren't running it on an raspbian, aren't you?
if yes, change your power supply.

brtwrs, 2019-03-07:
I dont think its the host. I use my stick on a normal pc and I don't see the "attempt power cycle". 
Most times I can get it to work again by replugging the stick and restarting zigbee2mqtt. But sometimes I can do reboots, replugs and restarts all I want but the stick wont respond. But when I reflash the stick everything works again.

nob0dy80, 2019-03-07:
@brtwrs 
well its just a thought.
had the same problems like you ...
did for sure 10 complete reinstalls. some times when the stick wont respond, i needed to reflash it completely to make it work again.

well finally its working here now.
my problem was that it seems like to be a bad idea to run it on a raspberry with 1gb RAM when all other adapters run on the same host.
using the raspberry now only for the zigbee and ble adapter in my flat, the rest runs on a virtual machine in the basement. since this time a never had zigbee-adapter problems again.

its only a guess...but maybe the adapter really gets in struggle when the RAM is insuffcient.

but i NEVER saw a power cycle ! 

you didnt answer if you are using a raspberry 3 or 3+! then its for sure the power supply!

brtwrs, 2019-03-07:
Thanks for your message but I use a 19" server in a rack so I dont think its a power or ram issue.. ;)

smarti2019, 2019-03-07:
i have the problems with the cc2531 on a raspberry pi and also with the intel nuc.
So it can not be a raspberry power problem.

Edzilla2000, 2019-03-07:
I have the exact same issues.
Sometimes it will work for a day or two, and sometimes it will work for minutes before it crashes.
I tried disabling power to the USB ports but that didn't help at all. The only thing that sometimes helps is unplugging it, waiting a while and replugging it.

I've tried it on a NUC, on a raspberry 3 and on a regurlar computer, I have the exact same issues.

nob0dy80, 2019-03-09:
i should have shut up ..after running fine 4 days. mine stopped working last night.
it didnt crash at all, it just stopped transfering.
 Zigbee publish to '0x0017880104781cce', genOnOff - on - {} - 11 failed with error Error: rsp error: 16

so what we can say till now is, its not host related cause it happens on different types of hosts, its not hardware stick related cause it happens on cc2531+cc2530 and it happens on different firmware versions (at least on cc2531 cause there are no dev version of cc2530 as far as i see).
this is going to be a teaser :-|

Koenkk, 2019-03-15:
@apeeters does it immediately crash after reflashing, or does it happen after a certain time?

apeeters, 2019-03-15:
@Koenkk Current observation seems to be: it varies: sometimes I can get it to work for a few seconds, other times immediate timeouts.

Koenkk, 2019-03-15:
@apeeters what about the reporting, is there a difference in having this disabled/enabled?

apeeters, 2019-03-15:
@Koenkk Reporting is currently disabled, I first want to get it stable again.

I seem to be able to get it working for a few seconds/minutes right after reflashing.

Koenkk, 2019-03-15:
@apeeters please try with the following firmware: https://drive.google.com/open?id=1-xzI6b8umZFpki-pfaKdLgcPrUUlswe5

Relative to the `20190223` firmware, it has an increased memory heap at the cost of direct connected devices to the coordinator (15 -> 5).

apeeters, 2019-03-15:
@Koenkk Currently running stable 🤞I'll let it run for a few more hours and then try to enable reporting again.

Koenkk, 2019-03-15:
@apeeters fingers crossed!

apeeters, 2019-03-18:
Still stable 🎉

Koenkk, 2019-03-18:
Also with reporting?

CMDR-Sloma, 2019-03-18:
I have had similar experience with previous version which kept crashing coordinator as soon as I was controlling groups with reporting on. Right now with this version from google drive it's super stable (reporting still on). The only minor issue is constant pinging of my two routers (CC2531 and cc2530+cc2591) but I can live with that ;). At the moment I have 33 different devices, 3 more routers (bulb) to hook up and 5 more sensors on the way. This is my ha favourite project.

nob0dy80, 2019-03-18:
@Koenkk 
can you do me a cc2530+cc2591 with the increased memory heap?  (or just tell me what do change, can compile myself)
5 direct connections should really be enough for everyone since every bulb is a router too

Koenkk, 2019-03-20:
@nob0dy80 

All `MAX_STABILITY` firmware flavours are available now: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator

nob0dy80, 2019-03-21:
@Koenkk 
awesome !
thanks

closed 2020-08-18 by github-actions[bot]