Koenkk/Z-Stack-firmware #48: Log: Reset coordinator Zigbee Stick blinkt rot (ioBroker)

Pascal51882, 2019-03-01:
Hallo, leider macht bei mir die neuste Firmware Probleme.
Ich habe ein paar Zigbee Sticks getestet. 2 mit denen es mit der alten Firmware (CC2531ZNP-Prod_20181024) funktioniert hat und einen neuen auf den ich nur die neuste (CC2531ZNP-Prod_20190109) aufgespielt habe. Sobald ich den serial Port hinterlege und der Adapter neustartet fängt der Zigbee Stick an rot zu blinken ab dem Log Punkt: Reset coordinator, mit der alten Firmware startet der Adapter normal durch und die LED geht aus bzw. leuchtet grün. (Ich kann zwar trotzdem keine Geräte verbinden, aber dieses Problem habe ich schon seit längerem auf ein mal...)

Hat noch jemand dieses Problem?

System:
rock64 mit OMV 4.X
ioBroker 1.4.2
Zigbee Adapter 0.9.2

nob0dy80, 2019-03-04:
Bitte englisch.

Pascal51882, 2019-03-05:
In English:

Hello, unfortunately the latest firmware has some problems. I have tested a few Zigbee sticks. 2 with the old firmware (CC2531ZNP-Prod_20181024) which has worked and a new on which has the newest (CC2531ZNP-Prod_20190109) on it. As soon as I enter the serial port and the adapter restarts, the Zigbee Stick starts flashing red from the log point: Reset coordinator, with the old firmware the adapter starts normally and the LED goes off or glows green. (Although I can not connect devices anyway, but I have this problem for a long time at a time ...) Does anyone else have this problem?

System:
rock64 with OMV 4.X
ioBroker 1.4.2
Zigbee Adapter 0.9.2

andreasbrett, 2019-03-05:
I have a CC2531 and also a Rock64 (though not with OMV but rather armbian). Could you try with the very latest firmware from Feb 23nd? It is from the development branch but works quite well for me:

https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/CC2531/bin

Attila-FIN, 2019-03-05:
I'm pretty new to ZigBee and CC2531, but I right away bought two pieces of CC2531. One I started to use with zigbee2mqtt the other I used for sniffing with TI Wireshark Packet Converter.
With the latest dev firmware, the CC2531 started to work very poorly, short distance reception (1-2 meters), often not even receiving anything. I thought that I could try the other CC2531 (which was flashed for packet capture), and re-flashed it with the same latest dev, and voilà, everything started to work perfectly. A startup I even saw some short term red lights coming from the CC2531, which I have never seen before on the first CC2531 dongle.
I don't know if the bad adapter worked better with previous firmware, I might do some tests with that adapter.
Anyway, it might be a bad quality CC2531 causing some issues.

Koenkk, 2019-03-05:
The short term red lights are normal during first initialization.

Pascal51882, 2019-03-11:
> The short term red lights are normal during first initialization.

Its longer than 10 Minutes (after 10 i gave up).
I now migrate my NAS to ESXi and will try it later again

Pascal51882, 2019-03-17:
With the newest version 23.02 the problem is solved. The red light is only for a very shot time and than it change to green.

closed 2019-03-17 by Pascal51882