Koenkk/Z-Stack-firmware #49: Lights can be turned off,but not on.

nob0dy80, 2019-03-03:
Hi,
after a new installation, i paired one of my hue lights.
I got a very strange problem ... i can turn hue lights off, but not on again over the data point 
coordinator is cc2530 with 20190109 (latest stable)

here is what the log says, cant find anything strange but maybe someone got an better eye on that.

`2019-03-03 17:40:34.309  - [34mdebug[39m: zigbee.0 Publish {"linkquality":92}
2019-03-03 17:40:36.675  - [34mdebug[39m: zigbee.0 User stateChange zigbee.0.001788010291b13b.state {"val":true,"ack":false,"ts":1551631236670,"q":0,"from":"system.adapter.admin.0","lc":1551631236670}
2019-03-03 17:40:36.687  - [34mdebug[39m: zigbee.0 publishFromState: deviceId=0x001788010291b13b, message={"cid":"genLevelCtrl","cmd":"moveToLevelWithOnOff","cmdType":"functional","zclData":{"level":0,"transtime":0},"cfg":{"manufSpec":0,"disDefaultRsp":0},"readAfterWriteTime":0}
2019-03-03 17:40:36.691  - [34mdebug[39m: zigbee.0 Zigbee publish to '0x001788010291b13b', genLevelCtrl - cmd moveToLevelWithOnOff - payload {"level":0,"transtime":0} - cfg {"manufSpec":0,"disDefaultRsp":0} - endpoint 11
2019-03-03 17:40:36.755  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"groupid":0,"clusterid":8,"srcaddr":31777,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":92,"securityuse":0,"timestamp":1053477,"transseqnumber":0,"len":5,"data":{"0":24,"1":36,"2":11,"3":4,"4":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":36,"cmdId":"defaultRsp","payload":{"cmdId":4,"statusCode":0}}}
2019-03-03 17:40:36.756  - [34mdebug[39m: zigbee.0 Publish {"linkquality":92}
2019-03-03 17:40:36.834  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"groupid":0,"clusterid":8,"srcaddr":31777,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":92,"securityuse":0,"timestamp":1053698,"transseqnumber":0,"len":13,"data":{"0":24,"1":37,"2":13,"3":1,"4":0,"5":0,"6":32,"7":1,"8":0,"9":33,"10":0,"11":64,"12":32},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":37,"cmdId":"discoverRsp","payload":{"discComplete":1,"attrInfos":[{"attrId":0,"dataType":32},{"attrId":1,"dataType":33},{"attrId":16384,"dataType":32}]}}}
2019-03-03 17:40:36.835  - [34mdebug[39m: zigbee.0 Publish {"linkquality":92}
2019-03-03 17:40:36.927  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"groupid":0,"clusterid":8,"srcaddr":31777,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":92,"securityuse":0,"timestamp":1053981,"transseqnumber":0,"len":19,"data":{"0":24,"1":38,"2":1,"3":0,"4":0,"5":0,"6":32,"7":1,"8":1,"9":0,"10":0,"11":33,"12":0,"13":0,"14":0,"15":64,"16":0,"17":32,"18":254},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":38,"cmdId":"readRsp","payload":[{"attrId":0,"status":0,"dataType":32,"attrData":1},{"attrId":1,"status":0,"dataType":33,"attrData":0},{"attrId":16384,"status":0,"dataType":32,"attrData":254}]}}
2019-03-03 17:40:36.927  - [34mdebug[39m: zigbee.0 Publish {"linkquality":92}
2019-03-03 17:40:56.979  - [34mdebug[39m: zigbee.0 Check online 0x001788010291b13b. 0x001788010291b13b
2019-03-03 17:40:57.033  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b is available
2019-03-03 17:40:57.034  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"available":true}
2019-03-03 17:40:57.035  - [34mdebug[39m: zigbee.0 Publish {"available":true}
2019-03-03 17:41:12.004  - [34mdebug[39m: zigbee.0 User stateChange zigbee.0.001788010291b13b.state {"val":false,"ack":false,"ts":1551631271998,"q":0,"from":"system.adapter.admin.0","lc":1551631271998}
2019-03-03 17:41:12.015  - [34mdebug[39m: zigbee.0 publishFromState: deviceId=0x001788010291b13b, message={"cid":"genLevelCtrl","cmd":"moveToLevelWithOnOff","cmdType":"functional","zclData":{"level":0,"transtime":0},"cfg":{"manufSpec":0,"disDefaultRsp":0},"readAfterWriteTime":0}
2019-03-03 17:41:12.021  - [34mdebug[39m: zigbee.0 Zigbee publish to '0x001788010291b13b', genLevelCtrl - cmd moveToLevelWithOnOff - payload {"level":0,"transtime":0} - cfg {"manufSpec":0,"disDefaultRsp":0} - endpoint 11
2019-03-03 17:41:12.083  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"groupid":0,"clusterid":8,"srcaddr":31777,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":89,"securityuse":0,"timestamp":1163872,"transseqnumber":0,"len":5,"data":{"0":24,"1":39,"2":11,"3":4,"4":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":39,"cmdId":"defaultRsp","payload":{"cmdId":4,"statusCode":0}}}
2019-03-03 17:41:12.084  - [34mdebug[39m: zigbee.0 Publish {"linkquality":89}
2019-03-03 17:41:12.151  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"groupid":0,"clusterid":8,"srcaddr":31777,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":89,"securityuse":0,"timestamp":1164088,"transseqnumber":0,"len":13,"data":{"0":24,"1":40,"2":13,"3":1,"4":0,"5":0,"6":32,"7":1,"8":0,"9":33,"10":0,"11":64,"12":32},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":40,"cmdId":"discoverRsp","payload":{"discComplete":1,"attrInfos":[{"attrId":0,"dataType":32},{"attrId":1,"dataType":33},{"attrId":16384,"dataType":32}]}}}
2019-03-03 17:41:12.152  - [34mdebug[39m: zigbee.0 Publish {"linkquality":89}
2019-03-03 17:41:12.218  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"groupid":0,"clusterid":8,"srcaddr":31777,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":89,"securityuse":0,"timestamp":1164290,"transseqnumber":0,"len":19,"data":{"0":24,"1":41,"2":1,"3":0,"4":0,"5":0,"6":32,"7":1,"8":1,"9":0,"10":0,"11":33,"12":0,"13":0,"14":0,"15":64,"16":0,"17":32,"18":254},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":41,"cmdId":"readRsp","payload":[{"attrId":0,"status":0,"dataType":32,"attrData":1},{"attrId":1,"status":0,"dataType":33,"attrData":0},{"attrId":16384,"status":0,"dataType":32,"attrData":254}]}}
2019-03-03 17:41:12.218  - [34mdebug[39m: zigbee.0 Publish {"linkquality":89}
2019-03-03 17:41:19.950  - [34mdebug[39m: zigbee.0 User stateChange zigbee.0.001788010291b13b.state {"val":true,"ack":false,"ts":1551631279945,"q":0,"from":"system.adapter.admin.0","lc":1551631279945}
2019-03-03 17:41:19.962  - [34mdebug[39m: zigbee.0 publishFromState: deviceId=0x001788010291b13b, message={"cid":"genLevelCtrl","cmd":"moveToLevelWithOnOff","cmdType":"functional","zclData":{"level":0,"transtime":0},"cfg":{"manufSpec":0,"disDefaultRsp":0},"readAfterWriteTime":0}
2019-03-03 17:41:19.967  - [34mdebug[39m: zigbee.0 Zigbee publish to '0x001788010291b13b', genLevelCtrl - cmd moveToLevelWithOnOff - payload {"level":0,"transtime":0} - cfg {"manufSpec":0,"disDefaultRsp":0} - endpoint 11
2019-03-03 17:41:20.043  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"groupid":0,"clusterid":8,"srcaddr":31777,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":89,"securityuse":0,"timestamp":1188748,"transseqnumber":0,"len":5,"data":{"0":24,"1":42,"2":11,"3":4,"4":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":42,"cmdId":"defaultRsp","payload":{"cmdId":4,"statusCode":0}}}
2019-03-03 17:41:20.044  - [34mdebug[39m: zigbee.0 Publish {"linkquality":89}
2019-03-03 17:41:20.128  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"groupid":0,"clusterid":8,"srcaddr":31777,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":86,"securityuse":0,"timestamp":1189018,"transseqnumber":0,"len":13,"data":{"0":24,"1":43,"2":13,"3":1,"4":0,"5":0,"6":32,"7":1,"8":0,"9":33,"10":0,"11":64,"12":32},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":43,"cmdId":"discoverRsp","payload":{"discComplete":1,"attrInfos":[{"attrId":0,"dataType":32},{"attrId":1,"dataType":33},{"attrId":16384,"dataType":32}]}}}
2019-03-03 17:41:20.129  - [34mdebug[39m: zigbee.0 Publish {"linkquality":86}
2019-03-03 17:41:20.200  - [34mdebug[39m: zigbee.0 Device 0x001788010291b13b incoming event:{"groupid":0,"clusterid":8,"srcaddr":31777,"srcendpoint":11,"dstendpoint":1,"wasbroadcast":0,"linkquality":89,"securityuse [... cut ...] ,"2":1,"3":0,"4":0,"5":0,"6":32,"7":1,"8":1,"9":0,"10":0,"11":33,"12":0,"13":0,"14":0,"15":64,"16":0,"17":32,"18":254},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":44,"cmdId":"readRsp","payload":[{"attrId":0,"status":0,"dataType":32,"attrData":1},{"attrId":1,"status":0,"dataType":33,"attrData":0},{"attrId":16384,"status":0,"dataType":32,"attrData":254}]}}
2019-03-03 17:41:20.200  - [34mdebug[39m: zigbee.0 Publish {"linkquality":89}`

nob0dy80, 2019-03-03:
ok, more funny: when i turn the light on again over power switch and i set the state "True" on the Switch state datapoint... the light goes off ! :-(

Koenkk, 2019-03-03:
What zigbee2mqtt version are you running?

dimaiv7, 2019-03-03:
I have the same problemm.

nob0dy80, 2019-03-03:
@Koenkk 
using iobroker with shepard 0.3.0. dont think zigbee2mqtt isnt involved.
something must have changed yesterday... cant be the firmware of the cc2530 and cant be the zigbee adapter for iobroker (last update 7 days ago)
my test installation yesterday worked ...and believe me, i did everything absolutely the same.

Koenkk, 2019-03-03:
@kirovilya can you help?

nob0dy80, 2019-03-03:
@Koenkk 
Maybe interesting too: If you add a Hue Dimmer Switch RWL021 it crashes completely:

| error | Caught by controller[0]: TypeError: Invalid cluster id: manuSpecificPhilips.
-- | -- | -- | --
| info | 1 (0x0017880103a540fc): 324131092621 - Philips Hue dimmer switch (EndDevice)

this was working yesterday too ! dont know if this is the right place for the issue cause it could be iobroker related too

nob0dy80, 2019-03-03:
@Koenkk 
i think i found the faulty one
https://github.com/ioBroker/ioBroker.zigbee/issues/218
https://github.com/Koenkk/Z-Stack-firmware/issues/49

Koenkk, 2019-03-03:
Looks like it, but then its indeed an iobroker issue.

closed 2019-03-03 by Koenkk