Koenkk/Z-Stack-firmware #51: IKEA remote and IKEA FLOALT

PJuergen, 2019-03-04:
I have a CC2530_CC2591with the version Coordinator firmware version: '20190109'
Installation:
L1527 - IKEA FLOALT with remote control E1524 - IKEA TRADFRI remote control paired. On / Off, brightness and light color can be controlled with the remote control.
Pairing the L1527 with the stick (6 x on/off) successful. On/Off, brightness and light color can now be controlled via mqtt2zigbee. With the remote control, the lamp can no longer be controlled.
Pairing the remote control with the CC2530 stick (4 x reset button next to the battery, entering the group key into the config) will succeed. Then pair the remote with the lamp. (Reset the lamp, repairing the remote control to the lamp successfully.)
The lamp can now be controlled via mqtt2zigbee. (On/off, brightness and light color)
The remote control now allows control of on/off and brightness.

Problem: The light color can not be changed with the remote control.

nob0dy80, 2019-03-04:
so... you can change colors in mqtt2zigbee but not with your directly paired remote....
clear thing..
this is not an firmware issue..
contact IKEA for help.

PJuergen, 2019-03-04:
The problem only occurs when both parts are paired with the stick.
Remote control and lamp work without pairing (reset remote, reset lamp, pairing lamp remote control) with the stick easily.

Koenkk, 2019-03-04:
What you should do here is only pair them to zigbee2mqtt (not to each other). Then find out the group of the remote control (http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#ikea-tradfri-remote-control-e1524) and add the FLOALT panel to that same group (http://www.zigbee2mqtt.io/information/groups.html)

PJuergen, 2019-03-07:
Hello,
I think it's a problem with the remote control software from IKEA.
Resetting and re-pairing the remote control and the lamp with the stick gives the same result.
Sniffing via Wireshark shows me only "unknown command" when pressing the color temperature button.

Is there a way to update the remote control software if there is one.

Koenkk, 2019-03-07:
@PJuergen this is expected, the arrow buttons use a TRADFRI specific command which can only be interpreted by TRADFRI devices.

capretto78, 2019-05-20:
> L1527

how can i pair the floalt panel withe the zigbee stick? 6x on off can't work for me... or i don't do it correctly. thanks

closed 2020-08-18 by github-actions[bot]