Koenkk/Z-Stack-firmware #52: Why is there never a router to router connection?

nob0dy80, 2019-03-09:
Hello,
maybe this is zigbee protocol related in general , but i wasnt able to find anything about it.

i noticed on the map,  that all router devices (lamps, power plugs) are always directed connected to the coordinator. i never see a connection between two routers, even that would make the link quality 100% better. 
A small snipped from my network map (green line is the way it should go to have a much better connection). 
[image]

Can anyone tell me something about that? Is that zigbee-standard related?

allofmex, 2019-04-11:
I think this can be closed, the latest iobroker network map shows sibling-links now.

nob0dy80, 2019-04-12:
@allofmex 
closing this cause of your expirience with that..

closed 2019-04-12 by nob0dy80