Koenkk/Z-Stack-firmware #54: More groups

nob0dy80, 2019-03-10:
@Koenkk 
I know you changed the group table size to 1 cause its not used Zigbee2mqtt anyway and cause of the heap. but....
Do you see a way to do a firmware with at least 2 groups too ?
Many people use your firmware (not only Zigbee2mqtt users) and i really dont want to change all my scripts and data points.
i also dont like that lights turn on one by one and not all at the same time like when they are in a group

kind regards.

Koenkk, 2019-03-10:
A group limit of 1 doesnt mean that there can only be one group in your zigbee network. You still can create as many as you want. The only limitation is that the coordinator can only be in one group. What is your use case for adding the coordinator to a group? (I dont see a use case for this)

nob0dy80, 2019-03-10:
@Koenkk 
before i had 2 groups .. in every group were 4 GU10 bulbs. so i just triggered the group states to turn on/off all 4 lights at the same time.
i always added only the bulps to a group (which are routers)... dont even know how to put a coordinator in a group.  Cant create a second group anyway. shown you the error message somewhere in a other issue. let me search for it here, cause im gone back to 20190109 where it still worked.

nob0dy80, 2019-03-10:
PS: Asked you the same on the other issue. seems like you overlooked it :-)
https://github.com/Koenkk/Z-Stack-firmware/issues/46#issuecomment-468680269

Koenkk, 2019-03-10:
IObroker shouldnt add the coordinator to groups, this is unnecessary. So it should be fixed there.

nob0dy80, 2019-03-10:
@Koenkk 
Thanks Koen, in that case i take the issue to iobroker.zigbee and refer to here!

Koenkk, 2019-03-10:
Great, closing this.

closed 2019-03-10 by Koenkk