Koenkk/Z-Stack-firmware #55: CC2530 + CC2591, where/what is SW1?

slackdance, 2019-03-10:
Simple question maybe, but I can't find a concrete answer. 

The first time I got this to send {"state": true}, I really have no idea how it happened. 

How do I trigger SW1 on the CC2530 + CC2591?

Koenkk, 2019-03-10:
@ptvoinfo could you help?

ptvoinfo, 2019-03-11:
SW1 is the hardware button on CC2531. Why do you need to trigger it? If you mean why the devices sent {"state": true} once, then it is ok after (re)flashing, because the internal memory is not initialized yet.

slackdance, 2019-03-11:
@ptvoinfo thanks for following up!

This is regarding the CC2530+CC2591 router (range extender in the guide). My CC2531 (coordinator) seems to be working fine.

Looking at older posts, it seemed that a powerloss set the device to "not routing", untill sw1 is triggered, and it reports {"state": true,"linkquality": xxx}. Have I misunderstood this, or have it been changed?

My zigbee network shows no connections to the router as it is now, with it sending state:false every minute. Before my powerloss, it was connected to almost every device, reporting state:true regularly. Initially, after first setup, it was reporting false - and my effort to set it to true did not give me a repeatable procedure of how to fix this, it has now been running since saturday, only connecting to the coordinator.

ptvoinfo, 2019-03-12:
@slackdance The router should report {"state": true} or {"state": false} every minute. It does not matter on CC2530 (on CC2531 this is the state of the green light). I don't know why other devices do not connect to the router in your case. It is possible they found a better path to the coordinator.

closed 2020-08-18 by github-actions[bot]