Koenkk/Z-Stack-firmware #58: Flashing CC2530 using Arduino+CCLib

GerSant, 2019-03-13:
I try to flash my new CC2530 using my Arduino UNO loaded with the CCLib Proxy and i obtain this meesage:

INFO: Found a CC2530 chip on COM5

Chip information:
      Chip ID : 0xa524
   Flash size : 256 Kb
    Page size : 2 Kb
    SRAM size : 8 Kb
          USB : No
Traceback (most recent call last):
  File "C:\Users\T15412\Downloads\CCLib-master\Python\cc_write_flash.py", line 48, in <module>
    hexFile.load()
  File "C:\Users\T15412\Downloads\CCLib-master\Python\cclib\cchex.py", line 128, in load
    self._loadHex()
  File "C:\Users\T15412\Downloads\CCLib-master\Python\cclib\cchex.py", line 367, in _loadHex
    raise IOError("Line %i: Unknown record type %02x" % (i, bType))
IOError: Line 12527: Unknown record type 05

What can be wrong?

Thanks

henriz, 2019-03-19:
This is a downside of the chosen way to program. In order to program your firmware, remove the second last text line from the hex file, it reads something like:
`:0400000500002D6367`

Then save the file, and reflash.

GerSant, 2019-03-19:
Thanks!!! Solved!

The Issue can be closed

closed 2020-08-18 by github-actions[bot]