Koenkk/Z-Stack-firmware #59: CC2531 shuts down soon after being plugged-in

suvl, 2019-03-13:
Hi. Out of the blue my cc2531 stopped working (it happened during the night, was not changing anything). The green led was off and `zigbee2mqtt` can't talk to the SoC.

Tried the usual: various plug-out and plug-in, reboots to the rpi, tried using it on the laptop, and tried flashing the latest `dev` as well as `master` firmware. Nothing.

Any idea on what might be goin' on? Shall I order a new one?

logs from `zigbee2mqtt`:

```
  zigbee2mqtt:info 3/13/2019, 3:51:02 PM Logging to directory: '/share/zigbee2mqtt/log/2019-03-13.15-51-02'
  zigbee2mqtt:info 3/13/2019, 3:51:03 PM Starting zigbee2mqtt version 1.2.1 (commit #unknown)
  zigbee2mqtt:info 3/13/2019, 3:51:03 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/13/2019, 3:51:03 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 3/13/2019, 3:52:03 PM Starting zigbee-shepherd
  zigbee2mqtt:error 3/13/2019, 3:52:03 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 3/13/2019, 3:52:03 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 3/13/2019, 3:52:03 PM Failed to start
	{"message":"Error: No such file or directory, cannot open /dev/ttyACM0123/hassio/addon/a0d7b954_ide","stack":"Error: Error: No such file or directory, cannot open /dev/ttyACM0123/hassio/addon/a0d7b954_ide"}
  zigbee2mqtt:error 3/13/2019, 3:52:03 PM Exiting...
```

nothing relevat from `dmesg`:

```
root@DietPi:~# dmesg -T | grep ACM 
[Tue Mar 12 11:37:38 2019] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device 
[Wed Mar 13 09:56:37 2019] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device 
```

nor from `journalctl`:

```
oot@DietPi:~# journalctl -xe --no-pager | grep ACM 
Mar 13 09:56:41 DietPi kernel: cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
```

any ideas? thanks!

Koenkk, 2019-03-13:
Please try reflashing the stick and provide the zigbee2mqtt debug log when it crashes again.

suvl, 2019-03-13:
@Koenkk flashed with latest `20190223` (using `cc-tool`).

Debug logs might not be that helpful:

```
[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-03-13T21:16:50: PM2 log: Launching in no daemon mode
2019-03-13T21:16:50: PM2 log: App [npm:0] starting in -fork mode-
2019-03-13T21:16:50: PM2 log: App [npm:0] online
> zigbee2mqtt@1.2.1 start /zigbee2mqtt-1.2.1
> node index.js
  zigbee2mqtt:info 3/13/2019, 9:16:53 PM Logging to directory: '/share/zigbee2mqtt/log/2019-03-13.21-16-53'
  zigbee2mqtt:debug 3/13/2019, 9:16:53 PM Removing old log directory '/share/zigbee2mqtt/log/2019-03-13.21-05-48'
  zigbee2mqtt:debug 3/13/2019, 9:16:54 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/share/zigbee2mqtt/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 3/13/2019, 9:16:54 PM Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:debug 3/13/2019, 9:16:54 PM Saving state to file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 3/13/2019, 9:16:54 PM Starting zigbee2mqtt version 1.2.1 (commit #unknown)
  zigbee2mqtt:info 3/13/2019, 9:16:54 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/13/2019, 9:16:55 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 3/13/2019, 9:17:55 PM Starting zigbee-shepherd
  zigbee2mqtt:error 3/13/2019, 9:17:55 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 3/13/2019, 9:17:55 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 3/13/2019, 9:17:55 PM Failed to start
	{"message":"Error: No such file or directory, cannot open /dev/ttyACM0123/hassio/addon/a0d7b954_ide","stack":"Error: Error: No such file or directory, cannot open /dev/ttyACM0123/hassio/addon/a0d7b954_ide"}
  zigbee2mqtt:error 3/13/2019, 9:17:55 PM Exiting...
```

liads, 2019-03-13:
Your serial port setting doesn't seem right..
Could it be you accidentally copy-pasted something extra into the config?
For the hassio addon it should look like:
```
  "serial_port": "/dev/ttyACM0",
```

suvl, 2019-03-14:
@liads it's right at the config:

```json
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": true,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://homeassistant",
  "serial_port": "/dev/ttyACM0",
  "devices": [],
  "network_key": [],
  "mqtt_user": <redacted>,
  "mqtt_pass": <redacted>,
  "log_level": "debug"
}
```
that path might be the way the mapping is done through docker and overlay, but I could not be certain.
Inside the container, the path checks out, it's not a symlink or something else.

suvl, 2019-03-14:
@Koenkk @liads just found out that you cannot take what's on the config json from hassio addon details: it might not be what's in the configuration.yaml at the `data_path`.

```yaml
serial:
  port: /dev/ttyACM0 # there was garbage here
```

Just wondering, at least when I added the `log_level` entry through the frontend, wouldn't it also change the `serial_port` key as it was not as stated in the json?

Nevertheless, the LED turning off was totally unrelated. Power save or something?

Thank you a lot for your help 👍

closed 2019-03-14 by suvl

liads, 2019-03-14:
When you start the addon the startup script loads the existing configuration.yaml, updates each setting defined in the addon json config, and rewrites the configuration.yaml. Not sure why it wouldn't update.

Regarding the LED turning off, I've seen it happen after a few minutes when I had the dongle plugged in without anything communicating with it.