Koenkk/Z-Stack-firmware #60: Devices > 30 ccznp has not been initialized yet

arteck, 2019-03-15:
i think we have a problem with the new Firmware.. 

here is my log from iobroker project.. the cc2531 stops .. with the message

zigbee.0	2019-03-15 15:50:53.566	error	at ZigbeeController.publish (/opt/iobroker/node_modules/iobroker.zigbee/lib/zigbeecontroller.js:366:20)
zigbee.0	2019-03-15 15:50:53.566	error	at Queue.(anonymous function) [as push] (/opt/iobroker/node_modules/queue/index.js:59:12)
zigbee.0	2019-03-15 15:50:53.566	error	at Queue.start (/opt/iobroker/node_modules/queue/index.js:145:17)
zigbee.0	2019-03-15 15:50:53.566	error	at queue.push (/opt/iobroker/node_modules/iobroker.zigbee/lib/zigbeecontroller.js:377:24)
zigbee.0	2019-03-15 15:50:53.566	error	at Endpoint.ep.functional (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-shepherd/lib/shepherd.js:590:25)
zigbee.0	2019-03-15 15:50:53.566	error	at ZShepherd._functional (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-shepherd/lib/shepherd.js:722:20)
zigbee.0	2019-03-15 15:50:53.566	error	at Object.af.zclFunctional (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-shepherd/lib/components/af.js:384:12)
zigbee.0	2019-03-15 15:50:53.566	error	at Object.af.send (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-shepherd/lib/components/af.js:106:16)
zigbee.0	2019-03-15 15:50:53.566	error	at Controller.request (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-shepherd/lib/components/controller.js:322:13)
zigbee.0	2019-03-15 15:50:53.566	error	at CcZnp.request (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/cc-znp/lib/ccznp.js:182:15)
zigbee.0	2019-03-15 15:50:53.566	error	Error: ccznp has not been initialized yet
zigbee.0	2019-03-15 15:50:53.565	error	uncaught exception: ccznp has not been initialized yet

here the completely log
[Neues Textdokument.TXT](https://github.com/Koenkk/Z-Stack-firmware/files/2971763/Neues.Textdokument.TXT)

any ideas ???

Koenkk, 2019-03-15:
Please try with the following firmware: https://drive.google.com/open?id=1-xzI6b8umZFpki-pfaKdLgcPrUUlswe5

Relative to the `20190223` firmware, it has an increased memory heap at the cost of direct connected devices to the coordinator (15 -> 5).

arteck, 2019-03-17:
reflash = new rebind the devices ??

arteck, 2019-03-18:
is it possible monitor the heap memory ??

Koenkk, 2019-03-18:
http://www.zigbee2mqtt.io/information/what_does_and_doesnt_require_repairing.html

Monitoring doesn't seem to be possible (not described here: http://dev.ti.com/tirex/content/simplelink_cc13x2_26x2_sdk_2_40_00_81/docs/zigbee/doxygen/zigbee/html/index.html)

arteck, 2019-03-20:
> 
> 
> Please try with the following firmware: https://drive.google.com/open?id=1-xzI6b8umZFpki-pfaKdLgcPrUUlswe5
> 
> Relative to the `20190223` firmware, it has an increased memory heap at the cost of direct connected devices to the coordinator (15 -> 5).

it works better..

closed 2020-08-11 by arteck