Koenkk/Z-Stack-firmware #62: INNR RS 128 T seen as 'undefined'

Eggn1n3, 2019-03-21:
Hi,
I just added an INNR tunable white GU10 spot but the logs shows:

 zigbee2mqtt:info 3/21/2019, 8:29:01 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-21.20-29-01'
  zigbee2mqtt:info 3/21/2019, 8:29:01 PM Starting zigbee2mqtt version 1.2.1 (commit #4048cb8)
  zigbee2mqtt:info 3/21/2019, 8:29:01 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/21/2019, 8:29:06 PM zigbee-shepherd started
  zigbee2mqtt:info 3/21/2019, 8:29:06 PM Coordinator firmware version: '20190223'

zigbee2mqtt:info 3/21/2019, 8:30:28 PM New device '**undefined**' with address 0x00158d0001c98215 connected!
  zigbee2mqtt:info 3/21/2019, 8:30:28 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0001c98215","meta":{}}'
  zigbee2mqtt:warn 3/21/2019, 8:30:28 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 3/21/2019, 8:30:28 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 3/21/2019, 8:30:28 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001c98215', payload '{"state":"ON","linkquality":52}'
  zigbee2mqtt:info 3/21/2019, 8:30:29 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001c98215', payload '{"state":"ON","linkquality":28,"brightness":254}'
  zigbee2mqtt:info 3/21/2019, 8:30:29 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001c98215', payload '{"state":"ON","linkquality":31,"brightness":254,"color_temp":370,"color_mode":2,"color":{"x":0.458,"y":0.41}}'
  zigbee2mqtt:info 3/21/2019, 8:30:59 PM Device incoming...
  zigbee2mqtt:info 3/21/2019, 8:30:59 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'

When I check https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js I see it listed.

Is there anything I should still do?

Thanks

Eggn1n3, 2019-03-21:
Update: same for INNR smartplug Sp 120:

zigbee2mqtt:info 3/21/2019, 8:47:09 PM New device 'undefined' with address 0x00158d00031eb064 connected!
  zigbee2mqtt:info 3/21/2019, 8:47:09 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d00031eb064","meta":{}}'
  zigbee2mqtt:warn 3/21/2019, 8:47:09 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 3/21/2019, 8:47:09 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 3/21/2019, 8:47:09 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00031eb064', payload '{"state":"OFF","linkquality":55}'
  zigbee2mqtt:info 3/21/2019, 8:47:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d00031eb064', payload '{"state":"OFF","linkquality":57,"power":0,"current":0,"voltage":0}'
  zigbee2mqtt:info 3/21/2019, 8:47:10 PM Successfully configured 0x00158d00031eb064 (0x00158d00031eb064)

Koenkk, 2019-03-21:
Somehow the pairing procedure hasnt been fully completed. can you try:

- stop zigbee2mqtt
- remove all entries of 0x00158d0001c98215  from database.db
- start zigbee2mqtt
- re-pair device

Eggn1n3, 2019-03-21:
I restarted zigbee2mqtt and now seen this in the log;

  zigbee2mqtt:info 3/21/2019, 9:31:34 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-21.21-31-34'
  zigbee2mqtt:info 3/21/2019, 9:31:35 PM Starting zigbee2mqtt version 1.2.1 (commit #4048cb8)
  zigbee2mqtt:info 3/21/2019, 9:31:35 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/21/2019, 9:31:35 PM zigbee-shepherd started
  zigbee2mqtt:info 3/21/2019, 9:31:35 PM Coordinator firmware version: '20190223'
  zigbee2mqtt:info 3/21/2019, 9:31:35 PM Currently 2 devices are joined:
  zigbee2mqtt:info 3/21/2019, 9:31:35 PM 0x00158d0001c98215 (0x00158d0001c98215): RS 128 T - Innr GU10 Spot 350 lm, dimmable, white spectrum (Router)
  zigbee2mqtt:info 3/21/2019, 9:31:35 PM 0x00158d00031eb064 (0x00158d00031eb064): SP 120 - Innr Smart plug (Router)

So, this is ok now, right?

Koenkk, 2019-03-21:
Looks good!

Eggn1n3, 2019-03-22:
I can confirm tht everything works as expected.
Sending **On** command works perfect:

> mosquitto_pub -h localhost -t 'zigbee2mqtt/0x00158d0001c98215/set'  -m '{"state":"ON"}

closed 2019-03-22 by Eggn1n3