Koenkk/Z-Stack-firmware #63: Firmware ZIP "Prod_20190223" not working

mf01, 2019-03-23:
The zip isn't working.

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2531/bin/CC2531ZNP-Prod_20190223.zip

Koenkk, 2019-03-23:
whats not working?

mf01, 2019-03-23:
The archive can't be opened.

[image]

Koenkk, 2019-03-23:
How did you download the file? whats the size, its working ok for me and many others.

mf01, 2019-03-23:
I have used this link "save link as..."

Version from Jan 9 is working.
```
-rw-rw-r-- 1 markus markus 332261 Feb  6 20:15 CC2531ZNP-Prod_20190109.zip
-rw-rw-r-- 1 markus markus  73075 Mar 23 19:55 CC2531ZNP-Prod_20190223.zip

```

Just tried another way and filesize is larger ... and zip is working.

```
-rw-rw-r-- 1 markus markus 332261 Feb  6 20:15 CC2531ZNP-Prod_20190109.zip
-rw-rw-r-- 1 markus markus 332011 Mar 23 20:14 CC2531ZNP-Prod_20190223.zip

```

Think it's my fault.... sorry

closed 2019-03-23 by Koenkk