Koenkk/Z-Stack-firmware #64: Can I completely rely on ESP8266?

2016for, 2019-03-31:
@Koenkk I want to be able to completely emulate the "multiple Xiaomi gateways" scenario.

- Can I create a coordinator with long range antenna device like CC2530 CC2591?

- Can I use CC2530 CC291 as an ESP8266 router? Basically place it anywhere in the house then.

- Also do xiaomi plugs keep on extending range?

- As in will this config work?
hub/coordinator ---- 30ft --- one xiaomi plug ---- 30ft ----- another plug .....

Background have a 4 story house with RPi Hassio in basement closet so coordinator and routers won't be able to talk to each other in Zigbee range. However each floor is fully wifi so interfacing zigbee router through wifi should work.

Would really appreciate your help.

Xeroxxx, 2019-06-24:
No. ESP8266 has only WiFi. You need Zigbee.

2016for, 2019-06-26:
@Xeroxxx I was hoping to add the CC2530/1 (for zigbee communication) to the ESP8266

Xeroxxx, 2019-06-26:
Technically it may be possible but I don't think anyone is going to write a driver for the ESP and I don't think it would fit to the flash.

There are toooo many other easier implementations like using it with a RPi. Only twice as big as an ESP and power consumption wise not really more. Do youself a favor and use a rpi.

JLFN, 2019-07-06:
> Technically it may be possible but I don't think anyone is going to write a driver for the ESP and I don't think it would fit to the flash.
>
> There are toooo many other easier implementations like [...]
Do you self a favor and dont talking nonsense.
http://www.zigbee2mqtt.io/how_tos/how_to_esp8266_with_cc2530.html

Xeroxxx, 2019-07-06:
Oh I understood he wants to use an ESP as CC2531/Coordinator.

However didn't know that. Thanks for clarification.

closed 2020-08-18 by github-actions[bot]