Koenkk/Z-Stack-firmware #65: What is the limit for NWK_MAX_DEVICE_LIST ?

swissbyte, 2019-04-02:
Hi 

I have tried to figure out where the limit for NWK_MAX_DEVICE_LIST comes from. 
There is a file OSAL_Nv.c is this responsible for saving all the Tables and other NV Informations?

Or in other words, does the Flash size limit the NWK_MAX_DEVICE_LIST because all bindings etc. are stored in the flash directly?

I know this is not an issue, but i dont know where else i should try to get in contact with you.

Would be very happy to get some answers. 
Thanks.

Koenkk, 2019-04-02:
Mostly because of `MAXMEMHEAP`, the bigger, the smaller the `NWK_MAX_DEVICE_LIST`.

swissbyte, 2019-04-03:
Im asking, because i would like to outsource the table to an external memory like an EEPROM. 
If i understand the User Manual for the Z-Stack correctly, everything which is responsible binding is stored in the NV-Memory. In case of the CC2531 it is the internal Flash

closed 2019-04-11 by swissbyte