Koenkk/Z-Stack-firmware #66: CC2530 + CC2591 silent, serial debug possible?

wunderbaum, 2019-04-05:
I flashed according to this [howto](http://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html)
But it does not appear in my network.
Unfortunately, this board has no LED to see any state of it.
How can I connect a serial adapter to see whats happening on the device?

Koenkk, 2019-04-07:
@ptvoinfo is this possible?

wunderbaum, 2019-04-07:
Meanwhile, I got it running. It seems to be necessary to power the device with a power supply and not via the SmartRF. But anyway, a serial debug could be helpful in many cases.

mabi, 2019-04-30:
I have the same issue, is there a way to debug?  
@wunderbaum what it the output of your power supply?  Thanks

wunderbaum, 2019-04-30:
Hi @mabi!
I now have a HLK-PM01 power supply with a small 5->3.3V Stepdown (https://www.aliexpress.com/item/Mini-DC-DC-12-24V-To-5V-3A-Step-Down-Power-Supply-Module-Voltage-Buck-Converter/32826540392.html).
I used those because I had them, I think its easier to use 3.3V version of the power supply.
Maybe there is a need for a capacitor to stabilize the current.

mabi, 2019-04-30:
Thank you @wunderbaum for the quick reply, I see the the output current of HLK-PM01 is approximately 1A so I will check if the power supply I am using can output that (I think so). I also bought but not yet installed the HLK-PM03 3.3V version, I will try using that.

MubiTec, 2019-05-07:
@wunderbaum: I tried to flash the cc2530 cc2591 even with this howto. But I did not really understadt the wiring. I did not really find the right pins at the cc2530 cc2591...

wunderbaum, 2019-05-07:
You should work alongside the howto and suddenly, its easy to find the right pins.
http://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

Then only connect the power to the pins to get it running.

closed 2020-08-18 by github-actions[bot]