Koenkk/Z-Stack-firmware #67: Provide hex and bin firmware files for 3.0.2

itsmepetrov, 2019-04-08:
I've tried to build own firmware from this description https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_3.0.x
but I've got a lot of warnings on one error:

```
Warning[Pe047]: incompatible redefinition of macro "ZDSECMGR_TC_DEVICE_MAX" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 8 
Warning[Pe047]: incompatible redefinition of macro  C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 9 
"DISABLE_GREENPOWER_BASIC_PROXY" 
Warning[Pe047]: incompatible redefinition of macro "CC2531ZNP" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 53 
Warning[Pe047]: incompatible redefinition of macro "NWK_MAX_DEVICE_LIST" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 54 
Warning[Pe047]: incompatible redefinition of macro "MAXMEMHEAP" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 56 
...
```

```
Error[e104]: Failed to fit all segments into specified ranges. Problem discovered in  
segment XDATA_N. Unable to place 2 block(s) (0x200 byte(s) total) in 0x1aa byte(s) of  
memory. The problem occurred while processing the segment placement command  
"-P(XDATA)XDATA_N=_XDATA_START-_XDATA_END", where at the moment of  
placement the available memory ranges were "XDATA:1d56-1eff" 
```

Could you please add hex and bin firmware files?

Koenkk, 2019-04-08:
Yes, but no publicly, there are still some issues that need to be solved first, please contact met on telegram (@koenkk)

creativtransfer, 2019-05-14:
There are no .bin files now at all. Please publish them or give us a hint how to convert the hex to bin. Does it work with hex2bin on linux?

dejavue1985, 2019-06-17:
hi there! zigbee-newbi here =) 
@Koenkk could you please send me the bin-file for the cc2531? my cc2531 is now "accidentially" running on V1.2HA and i want to bring it (back) to V3 but i don't have a CC programmer yet.
Tanks in advance!

Koenkk, 2019-06-17:
@dejavue1985 I will do this on the next updates of the firmware (first have to finish 1.2 firmwares).

closed 2020-08-18 by github-actions[bot]