Koenkk/Z-Stack-firmware #68: USB-Device not recognized

swissbyte, 2019-04-11:
Hi

I try to re-build the firmware with your patch. 
I have downloaded Z-StackHome1.2.2a.44539 and applied your patch to this directory. 
It patched me several files and i have checked that the patch was successfull. 

After that, i opened Projects\zstack\ZNP\CC253x\znp.eww
I changed the config from "Debug" to "ProdHex" and started a build. 
I've got a Hex-File wit 659kb size. I flashed it and... Nothing happens. 
Then after around 20-30 Seconds, windows tells me that there is an unrecognized USB-Device. 
If i flash your uploaded binary, everything works as espected. 

Could you please upload or send me a zip-file of your(Z-StackHome1.2.2a.44539) projectfolder?
I would be very happy. Then i can analyze why its not working here. 

If you dont want to share the Zip-File public, you could also upload it to my cloud: http://tinyurl.com/yyf2qel2 

Thank you so much!

Koenkk, 2019-04-11:
Have you also defined the symbol from step 6 https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/COMPILE.md ? 

What do you mean with nothing happens?

swissbyte, 2019-04-11:
Thank you for your fast answer. 

Yes, i have defined: 
FIRMWARE_CC2531
FIRMWARE_DEFAULT

I mean that the Dongle will not be recognized as an COM port. 
If i flash the hex file from your repo then the dongle gets recognized within a few seconds. 
With the hex file compiled by my own, the dongle does not get recognized.

swissbyte, 2019-04-11:
have you ever tested it with a USB Chip?

Koenkk, 2019-04-11:
What do you mean with USB chip? I can provide you the zip later.

swissbyte, 2019-04-11:
Hi 

I mean there are two different types of CC-Chips. One is the CC2530 which has only SPI or UART connectivity. The other one is the CC2531 which has additionaly a USB interface. 

I am using the CC2531 Chip. If i flash the chip with the regular untouched Firmware and plug the USB-Dongle (TI-Dev kit with CC2531) into the Host, then i get a virtual com-port to communicate with my device. 

If i apply your patch to the above mentioned "untouched" firmware and follow your guide to apply some defines and then flash this builded firmware to the chip, then the host doesnt recognize de USB-Device anymore. Why? I dont know. 

By the way, if i include #define INCLUDE_REVISION_INFORMATION in the preinclude.h then IAR throws me the following error: 

`Error[Pe020]: identifier "sblRev" is undefined C:\Texas Instruments\Z-StackHome1.2.2a.44539_mod\Components\mt\MT_SYS.c 538 `

The reason why i want to compile the firmware by my own is, because i want to be able to backup or clone a coordinator just in case something dies. For that, i must enable some compiler switches and rebuild the firmware. 

I also want to learn more about zigbee and your code. 
It seems that your project together with the zigbee-shepherd and zigbee2mqtt is the only one project with such functionalities. I would also like to support this project and contribute. Therefore i must go the hardway and learn how the Z-Stack is working... 

Thanks for your support!

swissbyte, 2019-04-11:
I have two USB-Dongles http://www.ti.com/graphics/tool/cc2531emk_800.jpg
And now i just pluged in the 2nd one instead of the first one and started zigbee2mqtt again. 
I was surprised, that all links were restores as if i had always the first dongle plugged in. 

As far as i know, the NWK Key is hardcoded in nwk_globals.c
Does this mean, that with your firmware, i would be able to connect to all homes which also uses your znp-firmware?

I thought that the Network will check the coordinators IEEE Address. Cause this address is different for my two dongles. There is also a frame_counter which could be checked. 
But for my two devices (Ikea tradfri lamp and remote) they just work as expected with a replacement coordinator.

Koenkk, 2019-04-11:
This is not the case for 1.2 (I'm currently working on 3.0 support, here this trick does not apply). Your statement is true, therefore users are advised to change their network key: http://www.zigbee2mqtt.io/how_tos/how_to_secure_network.html

swissbyte, 2019-04-11:
Thank you for your hint. 
Ok i see, i have searched at the wrong place. 
But im still interested to find out why i have such a lot of errors during compilation. 
So if it would be possible to get one complete zip-package i would be very happy. 

By the way, what would be the benefit to have Z-Stack 3.0 on coordinator side?
And one other question. Do you know if zigbee-shepherd or the coordinator does store some other informations beneath the Key? Are the routes also stored somewhere or will they be recreated everytime?

Thanks

Koenkk, 2019-04-11:
Could you contact me on telegram? @koenkk

swissbyte, 2019-04-11:
sure, thank you.

closed 2019-04-12 by swissbyte