Koenkk/Z-Stack-firmware #70: Totally crazy question

antst, 2019-04-13:
I don't have idea where else to ask, in fact.

One of strengths of zigbee is a "direct communication", like hue switch and hue lamps. It works instantly and even when hub is gone. Binding devices to devices.
But, once we go beyond, we found ourselves using multiple devices from multiple vendors, not all of them can use group comm or something else, etc, and then we go trough home automation hub.

And, for example, this is how chain looks in my bedroom (1st floor):
Xioami switch -> coordinator on the ground floor (maybe through a router) -> zigbee2mqtt -> mosquitto -> HA-> mosquitto -> zigbee2mqtt -> lamps (maybe through a router!).
This has too many links and involve "long distance" communication, which isn't perfect in the case of Hue either. All brings source of instability and lattency.

So, my idea, why don't we have special router firmware for CC2530/CC2531 (as they pretty much born for this function, taking into account price and everything), which will take part of basic automation. Like they can be paired to bulbs/switches locally, and/or send group messages, and listen for events and have few simple local rules, just in firmware. Kind of "proxy touch link device" or "basic automation offload engine".
Ideally it would be nice to configure them through zigbee also, but attaching ES8266 to them for sole purpose of remote configuration would be and option either.
So, then one put such "offload engines" in every room and have always locally reliable and quick setup for basic actions (like instanteneous on/off/toggle) and rest goes via HA.
This also makes it less comfortable if something will happen to HA hub.
Once story if you leave alone and hack you home automation, and another if you have big family full house of HA and your hub got some problems :)

So, then for simple button/sensor on/off/toggle action it would be local and quick, like I press Xiaomi switch, this "offload engine" will see the message and send group message to the lights.

I'd say for majority use cases, 2-5 rules would be more than enough for everybody.

What do you think?

Although, I am new to zigbee, I have pretty much 30 years of various programming experience and not complete stranger to hardware, and I am able and want to be part of development, testing, debugging etc.

But, at first, I definitely would like to hear opinion of someone who is more proficient with the topic.

slaapzakineentent, 2019-04-13:
Are you looking for the binding feature? https://www.zigbee2mqtt.io/information/binding.html

antst, 2019-04-13:
I am looking for the "local emulation" of this feature to the devices which can not be bind together.
You can't bind Xiaomi switch to tradfri lamp, for example.

antst, 2019-04-13:
No secret, that personally, my target are Xiaomi wired switches, which are (taking into account 2-gang switches and decoupled mode) are the most ingenious sensors I could dream of.

I am in the process to replace all wall switches in the house with those ones. And almost al 1-gand switches are replaced with 2-gang. Which gives me chance to control something extra in "usual" and non-intrusive way by using second button in decoupled mode.

Plus, in some cases, both buttons are decoupled as what 1-gang switch used to control, now filled with tradfri/hue bulbs, and now those two buttons allow me separately control 2 groups. While, in emergency case, I always can hold switch 10 sec to reset and get back direct control of lighting by power line.

ptvoinfo, 2019-04-14:
@antst This is possible in theory but requires many research an work.

closed 2020-08-18 by github-actions[bot]