Koenkk/Z-Stack-firmware #71: CC2530+CC2592 patch

kirovilya, 2019-04-20:
Hi, Koen.
I can not make proper PR to your firmware patch, but I have make patch to my-repo and get patch from it. Can you make correction to code for coordinator on Z-Stack 1.2.2 based on this patch ?
[cc2592-znp.zip](https://github.com/Koenkk/Z-Stack-firmware/files/3100543/cc2592-znp.zip)

It is for https://github.com/Koenkk/zigbee2mqtt/issues/1437 issue.

Koenkk, 2019-04-22:
This patch has quite a big diff from the current `firmware.patch`, how to apply it?

kirovilya, 2019-04-22:
I think - manually. It is not so much if look at file:
[image]

In one of the files - preinclude.h so many changes in patch, but in fact it is:
[image]

kirovilya, 2019-04-23:
Ok. I tried again [0001-cc2592.zip](https://github.com/Koenkk/Z-Stack-firmware/files/3108861/0001-cc2592.zip)

Koenkk, 2019-04-23:
That worked! Firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/max_stability/CC2530_CC2592

closed 2019-04-23 by kirovilya