Koenkk/Z-Stack-firmware #72: How to change the MAC address of coordinator?

he902, 2019-04-21:
I'm using CC2531 as the coordinator to add my smart Lock device into Home Assistant.
Currently everything works fine, except for remote Unlock failed (while using vendor's coordinator is OK to Unlock with the same PIN code).

I double the Lock device may have verified the MAC address of coordinator, and fail the Unlock request from any other coordinators not belong to the vendor.

How to change the MAC address of CC2531 coordinator? (so I'm able to crone the MAC)

Per the [TI doc](http://processors.wiki.ti.com/index.php/Z-Stack_FAQ#Device_MAC_Address_in_Z-Stack) it shall be able to change the MAC address of CC2531 by modifying firmware.

andreasbrett, 2019-04-21:
Please see the following doc on how to clone a cc2531. In that section you will find how to write a IEEE address (=MAC) of your choice to the CC2531.

https://github.com/andreasbrett/zigbee2mqtt.io/blob/patch-1/information/cc2531_backup_stick.md#how-to-clone-your-stick-windows-only-currently

he902, 2019-04-22:
@andreasbrett I followed your guidelines, and successfully written the new MAC into my CC2531 as Secondary MAC.
Now my CC2531 has two MAC:
Primary MAC: 0x00124b001936a4f9
Secondary MAC: 0x00124b0018eed067

But zigbee2mqtt still read the Primary MAC, here is the debug log:
```
  zigbee2mqtt:info 4/22/2019, 7:19:19 PM Starting zigbee2mqtt version 1.3.1 (commit #39d6a8e)
  zigbee2mqtt:info 4/22/2019, 7:19:19 PM Starting zigbee-shepherd
  zigbee2mqtt:info 4/22/2019, 7:19:25 PM zigbee-shepherd started
  zigbee2mqtt:info 4/22/2019, 7:19:25 PM Coordinator firmware version: '20190223'
  zigbee2mqtt:debug 4/22/2019, 7:19:25 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0xd067","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b001936a4f9","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190223},"startTime":1555931965,"joinTimeLeft":0}
```

And screenshots of using CC Debugger to read Primary/Secondary MAC from my CC2531:

[image]

[image]

Is there any config needed for zigbee2mqtt to enable using the Secondary MAC?

he902, 2019-04-24:
So current zigbee2mqtt only use the Primary MAC instead of Secondary MAC.
Is it a zigbee2mqtt bug or firmware bug?

he902, 2019-04-24:
After re-flashed the firmware, the zigbee2mqtt uses the Secondary MAC as expected now!
```
  zigbee2mqtt:info 4/24/2019, 11:38:21 PM Coordinator firmware version: '20190223'
  zigbee2mqtt:debug 4/24/2019, 11:38:21 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0xd067","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018eed067","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190223},"startTime":1556120301,"joinTimeLeft":0}
```

he902, 2019-04-24:
Per the [TI doc](http://processors.wiki.ti.com/index.php/Z-Stack_FAQ#Device_MAC_Address_in_Z-Stack), Z-Stack uses the following four-level hierarchy to determine the IEEE address that will be used for the device when it operates: 
    1. Read from Z-Stack non-volatile memory 
    2. Look-up from Secondary IEEE location 
    3. Look-up from Primary IEEE location 
    4. Create temporary using random number generation
Looks like the non-volatile memory on CC2531 has a copy of the Primary MAC --- who did it?
The re-flash of firmware has clean-up the non-volatile memory, thus goes to the 2nd level (the Secondary MAC).

Koenkk, 2019-04-25:
So why didn't it work the first time after reflashing?

he902, 2019-04-25:
The first time was not with reflashing, since my CC2531 had already with zigbee2mqtt firmware.
What I did in the first time was just to write MAC into the Secondary slot.
The second time, I did the reflashing.

he902, 2019-04-26:
So the `non-volatile memory` on CC2531 has a copy of the Primary MAC?

Curiously, which app did such copy? Is it by zigbee2mqtt?
If so, is it able to change the MAC by modifying `non-volatile memory` on CC2531 via zigbee2mqtt?
(without go through CC Debugger)

closed 2020-08-18 by github-actions[bot]

malosaa, 2022-08-03:
I have an issue to, well i have a new stick, using the last version of Z2M on home assistant.
I did not change something at all, only put old stick out and new in, and work as expected.

But here is the thing, i re-flashed the old coordinator as a router, and now when i want to paor it pairs, but as an coordinator.. so i did change the second IEEE but doesnt work, still keeps the first and still acting as a coordinator, even when the firmware is router firmware.

any advice ?

regards

Suschibomber, 2023-09-02:
> I have an issue to, well i have a new stick, using the last version of Z2M on home assistant. I did not change something at all, only put old stick out and new in, and work as expected.
>
> But here is [...]

Same for me... :(
Do you have a solution?

Hedda, 2024-01-16:
See related discussion here -> https://github.com/Koenkk/zigbee2mqtt/discussions/20724