Koenkk/Z-Stack-firmware #74: CC2530(+CC2591) crashes after few hours

jonnycastaway, 2019-04-22:
I dont know why nobody has the same questions but i go with zigbee2mqtt since months with various hardware and nothing is working longer than 1,5 days.

I firstly go with CC2531 but the range is a big problem. The max device firmware can't work with routers. The default and max stability firmware didn't work here with CC2530(+CC2591) as router very well.

The CC2530(+CC2591) as coordinator, default or max stability is equal,  crashes between 12h to 24h. With luck it's enough to reboot, else you have to reflash the whole CC2530(+CC2591).

I don't know if everything is wrong with my installation but i never get it constantly work more than 2 days in a row.

My actual installation is one CC2530(+CC2591) as coordinator (latest max stability firmware) at an Raspberry Pi 3+ connected over serial on GPIO. One CC2530(+CC2591) as router (latest router firmware). Zigbee2MQTT at latest master firmware. I checked latest dev firmware but it's the same.

4x Xiaomi Mija Temp/Humidity Sensors
1x Xiaomi Door/Windows Sensor
2x Philips Hue Outdoor Sensors
2x Osram Bulbs
7x Ikea Tradfri Bulbs
1x Ikea Tradfri Remote
4x Osram Smart+ Plugs

All works fine for 13 hours this day and then crash. Reboot dosn't work, i had to remove the power. And thats the best working setup i find the last months.

So what has to be done to get a stabilized working setup because my family is very sad (to say it friendly)

Koenkk, 2019-04-23:
This probably happens because running out of memory (CC2530 firmware has less memory available than the CC2531). 

In order to get things stable, I would recommend using the CC2531 with the max stability firmware as the coordinator and put the CC2530 + CC2591 close to it with a router firmware.

jonnycastaway, 2019-04-24:
Ok, thanks for your reply and tipps.
I'll test this improvement now.

Koenkk, 2019-04-24:
Let me know your findings.

jonnycastaway, 2019-04-24:
Oh, one question before i start.
I am right that every Router connects directly to the coordinator? If so i can't test your suggestion because i have a CC2530 + CC2591as Router Outdoor for the Philips Hue Outdoor Motion Sensors and it can't connect stable to the CC2531 thats indoor. Thats why i'll use the CC2530 + CC2591 as coordinator.

Koenkk, 2019-04-24:
Routers can also connect to other routers, they will automatically find the best connection.

Ormerod2, 2019-05-05:
Hi, same here with CC2530 + CC2590 (actually the new Ebyte E18). Unfortunately I don’t have any router since I am running a strong coordinator with 100mw. All my 12 Xiaomi devices have more than 70 linkquality but after 2-3days they stop working. I am running the dev coordinator 2019-4-25 from 1.2 Z-Stack.

jonnycastaway, 2019-05-10:
Hello @all,
after many days of testing my setup is now working without problems for days.
But the solution is not what you expect i think. All testing with several setups and @Koenkk  [solution](https://github.com/Koenkk/Z-Stack-firmware/issues/74#issuecomment-485936823) ended in problems. Devices that lost connection for hours or forever. Coordinator crashes and so on. But i found a solution for me.

I bought a new (from private person) Hue Bridge for 20€ and connected all Osram & Ikea Bulbs, the Hue Motion Sensors and the Osram Switches to it. My Osram Gardenpole that stopps working from one day also to another with zigbee2mqtt (only connects but no control) also (and he begins to functioning again).
Then i builded a second zigbee network with zigbee2mqtt (raspberri pi and one CC2530(+CC2591) over uart) and connetced the Xiaomi Sensors only to this network.
In Homeassistant i used the Hue Bridge Plugin additionally to mqtt and voila. All is running!
Since 6 Days without one single Problem. No extra Routers, no Linkqualitys that fluctuating between 0 and 120 and no broken connections.
My conclusion, either the Sticks are to small (Memory etc.) or the Rasoberry Pi is the limiting Factor.
I'm sorry that this is the naked truth but thats my conclusion after 3 Months of pain and an angry wife ;-)

Now all is superb!

allofmex, 2019-05-10:
@jonnycastaway 

>  Devices that lost connection for hours or forever

Your problem may not be a cc25xx crash but this here: [Enddevices unreachable because rejoin=false](https://github.com/Koenkk/zigbee2mqtt/issues/1474)

After I removed Osram and MüllerLicht, my net is stable too.
Please test without Osram routers, but with other routers. Ikea seems to be ok.

jonnycastaway, 2019-05-10:
@allofmex 
> Please test without Osram routers, but with other routers.

Mhm, eleminate the Osram Smart+ Plugs is not a solution for me. They switches my whole Media (TV, AVR, Subwoofer) and other things. And with the Hue Bridge there is no Problem.
The Ikea Router (i have one here) connects with zigbee2mqtt but i have never seen any single device (in the networkmap) that ever used it. The Xiaomi Sensors didn't work better (checked the Linkquality permanently) with this Router. So i removed it.
The biggest Problem i had with my Osram Gardenpole Light that works at the beginning for days (i think 5) and then went not controlable forever. All resets of Gardenpole, Zigbestick, Zigbe2mqtt whatever, nothing worked.

I forgot at the last Post, my Tradfri Remote is also connected to zigbee2mqtt and working perfect (Thanks Homeassistant that i can use all together).

allofmex, 2019-05-10:
@jonnycastaway 
> eleminate the Osram Smart+ Plugs is not a solution for me

Sure. We just need some more users to confirm that they cause problems with disconnecting enddevices. Then we can contact manufacturer.

andreasbrett, 2019-05-12:
I have 2 Osram Smart+ plugs in my network and every 3-4 weeks one of my IKEA tradfri bulbs gets disconnected. It's not always the same bulb. Didn't know that this might have to do with the Osram plugs. Looking at the logs though they behave different. Everytime I restart zigbee2mqtt those 2 plugs get "successfully configured" messages while none of my other devices shows this behaviour:

> 5/11/2019,` 4:40:58 PM - info: Successfully configured lr_power_outlet (0x7cb03eaa0*******)
> 5/11/2019, 4:41:00 PM - info: Successfully configured kt_power_outlet (0x84182600********)

(note: I masked the MAC addresses)

Regards
Andreas

closed 2020-08-18 by github-actions[bot]