Koenkk/Z-Stack-firmware #75: Can not compile firmware with provided patch.

antst, 2019-04-24:
I tried to compile CC2530 firmware with patch from dev branch and this is what I get:

```
Error[e16]: Segment XDATA_I (size: 0xa5 align: 0) is too long for segment definition. At least 0x56 more bytes needed. The problem occurred while processing the  
segment placement command "-Z(XDATA)XDATA_Z,XDATA_I=_XDATA_START-_XDATA_END", where at the moment of placement the available memory ranges  
were "XDATA:1eb1-1eff" 
   Reserved ranges relevant to this placement: 
   XDATA:1-4d0          XSTACK 
   XDATA:4d1-1eff       XDATA_Z 
   BIT:0-7              BREG 
   BIT:80-87            SFR_AN 
   BIT:90-97            SFR_AN 
   BIT:a0-af            SFR_AN 
   BIT:b8-c7            SFR_AN 
   BIT:e8-ef            SFR_AN 
Error while running Linker 

```

Koenkk, 2019-04-25:
What defines did you use?

antst, 2019-04-25:
```
FIRMWARE_CC2530
FIRMWARE_DEFAULT
```
And those were already there
```
FEATURE_SYSTEM_STATS
ASSERT_RESET
POWER_SAVING
FAKE_CRC_SHDW
xTC_LINKKEY_JOIN
```

Koenkk, 2019-04-25:
You should remove the ones that were already there.

antst, 2019-04-25:
It worked, thanks!

Koenkk, 2019-04-25:
Out of interest, why do you want to compile it?

antst, 2019-04-25:
As my CC2530 constantly hangs, I wanted  to test 57600 baud rate. It is either that, or  non-stop spamming from aqara wall switches by data requests (which coordinator doesn't answer,  attached capture log in #943) drives it crazy.

Koenkk, 2019-04-25:
Can you test if this has been fixed with the new firmwares (I just published them) https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin

antst, 2019-04-25:
What exactly? Hanging or non-answered data requests?

Koenkk, 2019-04-25:
Assuming you are now using the default cc2530 firmware, this one has far to less stack memory. This has been fixed in the above firmware.

antst, 2019-04-25:
I will try!
Found easy way to hang it :)
light.toggle for all 6 lights simultaneously hang it  for sure (

Koenkk, 2019-04-25:
Great, let me know!

antst, 2019-04-29:
This firmware seems to be stable and non-crashing. Works for 2 days already and I tried stuff which were giving me crash for granted, it passes now.
Thanks!

Koenkk, 2019-04-29:
Awesome!

danergo, 2019-11-13:
Hi, any chance to have a solution for the compilation problem below?

```
Fatal Error[e72]: Segment CRC_SHDW must be defined in a segment definition option (-Z, -b or -P) 
Error while running Linker 

```

- Workbench: 10.20.1
- Linker: C:\Texas Instruments\Z-Stack Home 1.2.2a.44539\Projects\zstack\Tools\CC2530DB\f8w2530.xcl
- Preprocessor macros defined: FIRMWARE_CC2530 && FIRMWARE_DEFAULT
- firmware.patch applied to Z-Stack Home 1.2.2a.44539
- "Depending if you want to compile the .hex or SBL firmware select - ProdHex or - ProdSBL" I have no idea where to chose from these

Thank you!

closed 2020-08-18 by github-actions[bot]

dkarthickch, 2021-01-20:
Hi I have used the HA 1.2 Source routing firmware in CC2531 and CC2530+CC2591 in 2 different hardware, initially i could able to add 30+ lighting modules (Zigbee 3.0), but later on I am unable to add any other modules, even after unpairing some modules I am unable to add further devices. Do you have any suggestion for this?