Koenkk/Z-Stack-firmware #77: CC2531 router has a very long device name

james-fry, 2019-04-25:
The name for my cc2531 routers flashed with [this f/w](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/CC2531/bin/CC2531_router_2018_09.zip) is very long:

> [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)

This is causing network map to get "corrupted", where the routers are very wide due to the name and other entities overprint:

[image]

Also it incorrectly states cc2530 rather than cc2531.

Obviously this is not a high priority defect, but next time a new bin is compiled please consider to fix these...

ptvoinfo, 2019-04-26:
@james-fry This problem is related to drawing, but not the router name in the firmware.

Koenkk, 2019-04-26:
@ptvoinfo yes that's one problem, the other problem is that the CC2530 router firmware has the same model as the CC2531 router firmware, therefore they cannot be distinguished.

james-fry, 2019-04-26:
> @james-fry This problem is related to drawing, but not the router name in the firmware.

@ptvoinfo I agree - the renderer for the map should handle a long name.
But I dont agree that the name needs to be that long.
> CC2531 Router
would be fine.
Even with a URL shortened link, maybe.
But no need for the length we see with the token "CC2530 router" there twice.
"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)"

Koenkk, 2019-04-26:
@james-fry this long name has nothing to do with @ptvoinfo firmware, but is a zigbee2mqtt bug.

james-fry, 2019-04-26:
> is a zigbee2mqtt bug.

Ah OK. Thx for letting me know.
Either way, all these are minor issues.

closed 2020-08-18 by github-actions[bot]