Koenkk/Z-Stack-firmware #78: router cc2531 does not connect

salopette, 2019-04-27:
Hallo, I flashed a cc2531 with router-cc2531-diag, but the red lamp flashes all the time only.
I see in zigbee2mqtt log synonymous nothing that a router has connected.

zibous, 2019-07-18:
Same problem here, when i flash the router-cc2531-std.hex:

Erase and flash the CC2531 works, so i hope :)
Press ** S2** - Short single press, but nothing happens - I see in zigbee2mqtt log synonymous nothing that a router has connected.

```
> cd flash_cc2531
> ./cc_chipid
 --->  ID = b524.

> cd ~/flash_cc2531  
> wget https://github.com/Koenkk/Z-Stack-firmware/raw/master/router/CC2531/bin/CC2531_router_2018_09.zip
> unzip CC2531_router_2018_09.zip 

## Erase and flash the CC2531 
> ./cc_erase
> ./cc_write router-cc2531-std.hex

[...]
ID = b524.
  reading line 12160.
  file loaded (12168 lines read).
writing page  96/ 96.
verifying page  96/ 96.
 flash OK.
[...]
```
sudo journalctl -u zigbee2mqtt.service -f

```
Jul 16 16:26:49 rb3 systemd[1]: Started zigbee2mqtt.
Jul 16 16:26:53 rb3 node[17777]:   zigbee2mqtt:warn 7/16/2019, 4:26:53 PM `permit_join` set to  `true` in configuration.yaml.
Jul 16 16:26:53 rb3 node[17777]:   zigbee2mqtt:warn 7/16/2019, 4:26:53 PM Allowing new devices to join.
Jul 16 16:26:53 rb3 node[17777]:   zigbee2mqtt:warn 7/16/2019, 4:26:53 PM Set `permit_join` to `false` once you joined all devices.
Jul 18 11:36:44 rb3 systemd[1]: Stopping zigbee2mqtt...
Jul 18 11:36:44 rb3 systemd[1]: Stopped zigbee2mqtt.
Jul 18 11:36:44 rb3 systemd[1]: Started zigbee2mqtt.
Jul 18 11:36:49 rb3 node[19622]:   zigbee2mqtt:warn 7/18/2019, 11:36:49 AM `permit_join` set to  `true` in configuration.yaml.
Jul 18 11:36:49 rb3 node[19622]:   zigbee2mqtt:warn 7/18/2019, 11:36:49 AM Allowing new devices to join.
Jul 18 11:36:49 rb3 node[19622]:   zigbee2mqtt:warn 7/18/2019, 11:36:49 AM Set `permit_join` to `false` once you joined all devices.
.....
```
[image]

abmantis, 2019-09-30:
Also having this issue when trying to pair with ZHA. Any tips?

zibous, 2019-10-01:
@abmantis 

I bought OSRAM Smart + plug's and now use them as a router. These have a greater range and it works with connecting to the coordinator and the other zigbee devices (Aqara door, temp, waterleak, motion devices)

abmantis, 2019-10-01:
So, after leaving the network open for a lot of time (like 15 minutes), it finally joined!

cconde, 2019-12-02:
I'm having the same problem, Is th 15 minutes a common solution?

cconde, 2019-12-02:
Seems waiting is not a solution for me... Unable to connect so far

ptvoinfo, 2019-12-03:
@cconde If you cannot pair the CC2530 router:
1. Ensure that only one network around accepts joining. 
2. Move the outer closely to the coordinator ~1m.
3. Power on the router, wait 2 seconds, power off. Repeat this cycle three times (this producedure makes full reset).

cconde, 2019-12-13:
Done all three unsuccessfully.

Zigbee sniffing shows coordinator sends an join successful packet and the transport key and at that point router reboots and tries again to join.
I'm suspecting router can't decode Transport Key properly as Philips encodes it with a non standard Commissioning Key.

closed 2020-08-18 by github-actions[bot]