Koenkk/Z-Stack-firmware #79: CC2530 + CC2592 BDB_START_COMMISIONING (0x02 No Network)

tienpjo, 2019-05-06:
I am using cc2530 - ebyte (E18-MS1PA1-IPX) module, I used ZNP-ZStack 3.0.2 from Ti, define symbol HAL_PA_LNA_CC2592 in precesser of IAR, but when installing a zigbee network (I follow https: //sunmaysky.blogspot.com/2017/02/use-ztool-z-stack-30-znp-to-set-up.html#comment-form). When Joining router enters the network with the command APP_CNF_BDB_START_COMMISSIONING (0x2F05)
     CommissioningMode: (0x02) Network Steering (0x2)
It responds to me as No network (0x02). Please guide me through this error!
thanks you very much !!
sorry: sorry with my English

Koenkk, 2019-05-06:
In combination with what software are you using it? You can use the init procedure from https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L258

tienpjo, 2019-05-06:
> In combination with what software are you using it? You can use the init procedure from https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L258

can you guide me I don't really understand, I'm working with CC2530ZNP
thank you very much!

Koenkk, 2019-05-06:
If you want to know what these commands are exactly doing, please refer to the Z-Stack documentation (included when installing Z-Stack 3.0.2 on your PC) or the TI forum (https://e2e.ti.com/)

closed 2020-08-18 by github-actions[bot]