Koenkk/Z-Stack-firmware #80: Communication CC2530 Router to Arduino via UART

MudiStar, 2019-05-10:
Hello everyone,

I want to connect an Arduino to a cc2530 Router over the UART interface (RX/TX).
Is the pinout the same like the coordinator software?
Coordinator:
P02 - RX
P03 - TX

The goal is to send commands to the Router and it should forward it to the Arduino.
I searched a lot, but I couldn’t find an answer.

Thanks for answering my question

MudiStar, 2019-05-14:
I connected the Arduino to the CC2530 router like it is described in the coordinator article and I got a status message (Baud Rate: 115200) when I stop pulling the rest pin to ground.
But I couldn’t receive a mqtt message. 
Does someone know where I can find the source code?

ptvoinfo, 2019-05-14:
@MudiStar The CC2530 router does not send anything useful (MQTT) for you to the UART port. Please, read more about the router role here:
https://www.zigbee2mqtt.io/information/zigbee_network.html

MudiStar, 2019-05-14:
Hmm I'm not asking a top-level question, so this doesn't help.

A Hue light bulb works as repeater and I can also send and receive data. So in general a router on CC2530 basis can forward data to UART.

Like this project but here is no source code provided:
http://wsn.blogs.ua.sapo.pt/970.html

ptvoinfo, 2019-05-14:
@MudiStar It is possible only if you'll write a new firmware.

MudiStar, 2019-05-15:
Ok that’s what I feared

Is here someone who is capable of doing this? 
I found some other topics where people also would like to have this functionality for their smart home 

I mean maybe it is possible to work out a deal 😉 
In the end I would like to provide the .hex for everyone

SAGARP169, 2019-09-26:
@Mudistar did you find anything on this, i am too trapped in the same condition now

MudiStar, 2019-09-26:
I did a lot of research

The closest I got was the firmware of this module:

https://www.waveshare.com/wiki/Core2530_(B)

You can flash the router firmware provided on any Cc2530 module

It joined my network without problems 

I then used Wireshark to see what happens to the message I have sent from a ftdi to the module via uart 

The message gets somehow through but looks different like e.g. from a xiaomi power plug.
But the real problem is, that the module causes a lot of traffic in the network marked as "malformed Paket" even in idle 

I'm on vacation atm so I don't have pictures of this right now

I wrote a mail to waveshare but they obviously won't adapted the code
At least I tried

Maybe we should try to crowedfound this code 😉

SAGARP169, 2019-09-28:
@Mudistar Thank you!
your solution did work for me!
As i am not at all good in programming i was looking for this type of ready firmware.
What i wanted is my own zigbee mesh network, and its done.
Though i have only created mesh network of 3 modules as i have only 3 of them.
I will test it further after adding extra modules.
Thanks !

SAGARP169, 2019-09-28:
RX P1.4
TX P1.5
BAUDRATE  38400

MudiStar, 2019-09-29:
Ok I see

You were just looking for a stand alone solution without integration in a system with a raspberry

I'm glad that I could help you 👍

closed 2020-08-18 by github-actions[bot]