Koenkk/Z-Stack-firmware #87: CC2531_2019045: zigbee2mqtt:error 2019-5-19 19:38:39 Failed to start

Smiggel, 2019-05-19:
I flashed my stick to firmware CC2531_2019045. After doing that, I noticed my zigbee2mqtt is not working anymore. I get this errors:

> zigbee2mqtt:error 2019-5-19 19:38:39 Error while starting zigbee-shepherd!
mei 19 19:38:39 Domoticz-VPN npm[1439]:   zigbee2mqtt:error 2019-5-19 19:38:39 Press the reset button on the stick (the one closest to the USB) and start again
mei 19 19:38:39 Domoticz-VPN npm[1439]:   zigbee2mqtt:error 2019-5-19 19:38:39 Failed to start
mei 19 19:38:39 Domoticz-VPN npm[1439]:         {"message":"rsp error: 2","stack":"Error: rsp error: 2\n    at rspHdlr (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/controller.js:309:29)\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:258:13)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at CcZnp._mtIncomingDataHdlr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:360:14)\n    at Immediate._onImmediate (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:336:22)\n    at runCallback (timers.js:705:18)\n    at tryOnImmediate (timers.js:676:5)\n    at processImmediate (timers.js:658:5)"}

When I go back to firmware CC2531_20190223 it all runs fine again. I Run Zigbee2mqtt on the latest Domoticz version 4.10717

Koenkk, 2019-05-19:
Could you share some details about your network? How many devices? Router/endevices?

Smiggel, 2019-05-19:
Ofcourse. I just realize I still use zigbee2mqtt 1.1.1. Perhaps that is the problem.

I run Domoticz and Zigbee2Mqtt on a Raspberry Pi 3B+. I have 2 Xiaomi Aqara PIR sensors, 5 Xiao mi aqara Smart Temperatuur Hu Mi dity sensors and 2 Xiaomi Aqara Raam Deur Sensors. So, not that much actually. Next to that I have two CC2531 sticks flashed as a router. 

The network is stable on the old firmware. On the new it crashes during start. But perhaps that has got to do with the zigbee2mqtt version?

Koenkk, 2019-05-19:
Maybe, please update to the latest zigbee2mqtt, reflash the firmware and try again.

Smiggel, 2019-05-19:
New Zigbee2mqtt version doesn't make a difference. Still same error.

Smiggel, 2019-05-19:
Restored my Domoticz image and went back to version 1.1.1. 1.4 is giving my problems. Routers drop from the network. See issue https://github.com/Koenkk/zigbee2mqtt/issues/1536

closed 2020-08-18 by github-actions[bot]