Koenkk/Z-Stack-firmware #88: Somehow I soft-bricked the coordinator?

lucaspiller, 2019-05-21:
So last night I was playing around setting up zigbee devices, and after turning off (from the physical switch) a group of lights everything stopped working. I've tried restarting zigbee2mqtt, resetting the coordinator (from the button on the device), and even tested it on a different machine but I get the same results. This is what I get when booting zigbee2mqtt with the debug flag:

```
  zigbee2mqtt:info 5/21/2019, 7:07:45 AM Logging to directory: '/app/data/log/2019-05-21.07-07-45'                                                                                                                               [35/73]
  zigbee2mqtt:debug 5/21/2019, 7:07:45 AM Removing old log directory '/app/data/log/2019-04-10.22-04-11'                                                                                                                                
  serialport/bindings loading LinuxBinding +0ms                                                                                                                                                                                         
  zigbee2mqtt:debug 5/21/2019, 7:07:45 AM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackup
Path":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'                                                                                                                                                      
  zigbee2mqtt:debug 5/21/2019, 7:07:45 AM Loaded state from file /app/data/state.json                                                                                                                                                   
  zigbee2mqtt:debug 5/21/2019, 7:07:45 AM Saving state to file /app/data/state.json                                                                                                                                                     
  zigbee2mqtt:info 5/21/2019, 7:07:45 AM Starting zigbee2mqtt version 1.4.0 (commit #unknown)                                                                                                                                           
  zigbee2mqtt:info 5/21/2019, 7:07:45 AM Starting zigbee-shepherd                                                                                                                                                                       
  zigbee-shepherd:init zigbee-shepherd booting... +0ms                                                                                                                                                                                  
  cc-znp Initialize, opening serial port +0ms                                                                                                                                                                                           
  serialport/stream opening path: /dev/ttyACM0 +0ms                                                                                                                                                                                     
  serialport/binding-abstract open +0ms                                                                                                                                                                                                 
  serialport/stream _read queueing _read for after open +3ms                                                                                                                                                                            
  serialport/bindings/poller Creating poller +0ms                                                                                                                                                                                       
  serialport/stream opened path: /dev/ttyACM0 +151ms                                                                                                                                                                                    
  serialport/stream _write 1 bytes of data +1ms                                                                                                                                                                                         
  serialport/binding-abstract write 1 bytes +156ms                                                                                                                                                                                      
  serialport/stream _read reading +2ms                                                                                                                                                                                                  
  serialport/binding-abstract read +2ms                                                                                                                                                                                                 
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms                                                                                                                                                     
  serialport/bindings/unixRead Starting read +0ms                                                                                                                                                                                       
  serialport/bindings/unixWrite write returned null 1 +7ms                                                                                                                                                                              
  serialport/bindings/unixWrite wrote 1 bytes +0ms                                                                                                                                                                                      
  serialport/bindings/unixWrite Finished writing 1 bytes +0ms                                                                                                                                                                           
  serialport/stream binding.write write finished +10ms                                                                                                                                                                                  
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +8ms                                                                                                                                                        
  serialport/bindings/poller Polling for "readable" +17ms                                                                                                                                                                               
  cc-znp The serialport /dev/ttyACM0 is opened. +1s                                                                                                                                                                                     
  zigbee-shepherd:request REQ --> SYS:version +0ms                                                                                                                                                                                      
  cc-znp:SREQ --> SYS:version, {} +0ms                                                                                                                                                                                                  
  serialport/stream _write 5 bytes of data +1s                                                                                                                                                                                          
  serialport/binding-abstract write 5 bytes +1s                                                                                                                                                                                         
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +1s                                                                                                                                                      
[... cut ...]
  zigbee2mqtt:info 5/21/2019, 7:07:53 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
```

The only thing I haven't tried is reflashing the firmware (I bought the CC2531 coordinator ready flashed with a 201902?? firmware), which I will try this evening. I assume something has become messed up on the memory of the device. Is this not what the 'Reset' button is supposed to fix though?

As a side note, is there any way to set my lightbulbs as end devices rather than routers?

Koenkk, 2019-05-21:
It seems that indeed something has been messed up in the memory, reflashing should fix it.

AFAIK in Zigbee it's not possible to change devices from routers to end devices.

lucaspiller, 2019-05-21:
Yup reflashing fixed it :+1:

closed 2019-05-21 by lucaspiller