Koenkk/Z-Stack-firmware #89: Z-Stack 3.0 for CC2530

jiki974, 2019-05-21:
Please, can you provide instructions for building Z-Stack 3.0 for CC2530?

Koenkk, 2019-05-23:
Available now: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin can you let me know if it works?

jiki974, 2019-05-23:
Thank you so much for your incredible work ! It works fine but sometimes I need to restart my CC2530 board in order to made it work. However I have a problem with zigbee green power devices, their didn't appears ... I thought that Zigbee 3.0 is compatible with Zigbee green power.

Koenkk, 2019-05-23:
Yes it is, but some additional things are needed to support this (green power commissioning).

jiki974, 2019-05-23:
Great reactivity thank you !

closed 2019-05-23 by jiki974