Koenkk/Z-Stack-firmware #91: "rsp error 2" using CC2531_20190425

bl4ckwhiteeagle, 2019-05-27:
I've upgraded and Formatted my CC2531 with a USB-Flasher CC2531 an getting "rsp error 2" using CC2531_20190425.zip

Older Z-Stack firmware works, almost without any problems (except Xiaomi Sensors/Switches)

Tried 3 different CC2531, reflashed old firmware an it works out of box again, only the new Firmware makes this problem.

Very rar is this error message/a reconnect resolves this problem, but i get this kernel message with the older stack-firmware, too:
cdc_acm 1-2.1:1.0: failed to set dtr/rts

Hardware:
Intel NUC7I3BNH with ESXI 6.7 Virtualization (Works like a charm :)))))) )
Latest IOB (obroker update & upgrade & upload all)
Zigbee Version 0.10.3 in /opt/iobroker/node_modules/iobroker.zigbee
node: v8.15.1
NPM: 6.4.1

Any ideas?
I'll try now on bare system without Virtualisation with a clone of /opt/iobroker

Many thanks!

zigbee.0 | 2019-05-27 22:01:20.400 | debug | User  stateChange zigbee.0.info.connection  {"val":false,"ack":false,"ts":1558987280396,"q":0,"from":"system.adapter.zigbee.0","user":"system.user.admin","lc":1558986752062}
-- | -- | -- | --
zigbee.0 | 2019-05-27 22:01:20.398 | debug | User  stateChange zigbee.0.info.pairingMessage {"val":"Error: Error while  starting zigbee-shepherd!. Error: rsp error:  2","ack":false,"ts":1558987280393,"q":0,"from":"system.adapter.zigbee.0","user":"s
zigbee.0 | 2019-05-27 22:01:20.394 | error | Error while starting zigbee-shepherd!. Error: rsp error: 2
zigbee.0 | 2019-05-27 22:01:14.847 | info | Starting zigbee-shepherd
info.0 | 2019-05-27 22:00:23.701 | info | Popup-News readed...
zigbee.0 | 2019-05-27 22:00:14.837 | info | Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
zigbee.0 | 2019-05-27 22:00:06.049 | info | Reset coordinator
zigbee.0 | 2019-05-27 22:00:06.049 | info | Queue is: true
zigbee.0 | 2019-05-27 22:00:06.049 | info | Start on port: /dev/ttyACM0 with panID 6754 channel 25
zigbee.0 | 2019-05-27 22:00:06.048 | info | starting. Version 0.10.3 in /opt/iobroker/node_modules/iobroker.zigbee, node: v8.15.1
host.iob | 2019-05-27 22:00:01.698 | info | instance system.adapter.zigbee.0 started with pid 16014
host.iob | 2019-05-27 21:59:59.183 | info | instance system.adapter.zigbee.0 terminated with code 0 (OK)
host.iob | 2019-05-27 21:59:59.162 | info | stopInstance system.adapter.zigbee.0 killing pid 15854
host.iob | 2019-05-27 21:59:59.162 | info | stopInstance system.adapter.zigbee.0
host.iob | 2019-05-27 21:59:59.162 | info | object change system.adapter.zigbee.0
zigbee.0 | 2019-05-27 21:59:59.131 | debug | statesDB connected
zigbee.0 | 2019-05-27 21:59:59.116 | debug | objectDB connected
host.iob | 2019-05-27 21:59:57.933 | info | "system.adapter.zigbee.0" disabled
host.iob | 2019-05-27 21:59:57.933 | info | object change system.adapter.zigbee.0
host.iob | 2019-05-27 21:59:57.884 | info | instance system.adapter.zigbee.0 terminated with code 0 (OK)
host.iob | 2019-05-27 21:59:56.229 | info | stopInstance system.adapter.zigbee.0 killing pid 15348
host.iob | 2019-05-27 21:59:56.229 | info | stopInstance system.adapter.zigbee.0
zigbee.0 | 2019-05-27 21:59:56.230 | info | zigbee-shepherd stopped
host.iob | 2019-05-27 21:59:56.229 | info | object change system.adapter.zigbee.0
admin.0 | 2019-05-27 21:59:10.882 | info | successful connection to socket.io from ::ffff:192.168.0.150
zigbee.0 | 2019-05-27 21:58:02.690 | error | Error while starting zigbee-shepherd!. Error: rsp error: 2

bl4ckwhiteeagle, 2019-05-27:
Same error on a bare metal system, seems to be a firmware related problem. USB Extenstion Cable also removet/same error :-/ switching back to older firmware again on main system until this error is resolved

Koenkk, 2019-05-28:
Can you check if it works with zigbee2mqtt 1.4.0?

bl4ckwhiteeagle, 2019-05-28:
Short answear: Yes I see there device with Coordinator firmware version: '20190425' :)

Plugged in CC2531 in hardware based Test-Server directly

Upgraded nodejs to v10.15.3 as of zigbee2mqtt requied that...

Cloned/Configured zigbee2mqtt

CC2531 green light is of / zigbee2mqtt request timed out, a reconnect of the device got it working with it

1
###
  zigbee2mqtt:info 2019-5-28 22:23:16 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-28.22-23-16'
  zigbee2mqtt:info 2019-5-28 22:23:16 Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
  zigbee2mqtt:info 2019-5-28 22:23:16 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-5-28 22:23:27 Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:error 2019-5-28 22:23:31 Not connected to MQTT server!
  zigbee2mqtt:error 2019-5-28 22:23:31 Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
  zigbee2mqtt:info 2019-5-28 22:23:31 zigbee-shepherd stopped
###

2
###
  zigbee2mqtt:info 2019-5-28 22:24:08 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-28.22-24-08'
  zigbee2mqtt:info 2019-5-28 22:24:08 Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
  zigbee2mqtt:info 2019-5-28 22:24:08 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-5-28 22:24:22 zigbee-shepherd started
  zigbee2mqtt:info 2019-5-28 22:24:22 Coordinator firmware version: '20190425'
  zigbee2mqtt:info 2019-5-28 22:24:22 Currently 0 devices are joined:
  zigbee2mqtt:warn 2019-5-28 22:24:22 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-5-28 22:24:22 Allowing new devices to join.
  zigbee2mqtt:warn 2019-5-28 22:24:22 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-5-28 22:24:22 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-5-28 22:24:22 Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 2019-5-28 22:24:22 zigbee-shepherd ready
###
  
iobroker zigbee didn't come up as of the error message:
Error: The module '/opt/iobroker/node_modules/iobroker.zigbee/node_modules/serialport/build/Release/serialport.node'

So i did a:
sudo -u iobroker npm rebuild
Got the error:
error Failed at the unix-dgram@0.2.3 install script.

Did:
sudo -u iobroker npm install -d unix-dgram

iobroker and zigbee module starts without any problems but i get the rsp error again :-/
Error while starting zigbee-shepherd!. Error: rsp error: 2

I'll try by reinstalling/reconfiguring the zigbee adapter

bl4ckwhiteeagle, 2019-05-28:
No succes, a fresh reinstall in a new directory /opt/iobroker and zigbee adapter does also the error:
Error while starting zigbee-shepherd!. Error: rsp error: 2

/opt/iobroker/iobroker stop

and 

cd /opt/zigbe2mqtt
npm start

works without any problems again:
  zigbee2mqtt:info 2019-5-28 23:21:41 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-28.23-21-41'
  zigbee2mqtt:info 2019-5-28 23:21:41 Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
  zigbee2mqtt:info 2019-5-28 23:21:41 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-5-28 23:21:53 zigbee-shepherd started
  zigbee2mqtt:info 2019-5-28 23:21:53 Coordinator firmware version: '20190425'
  zigbee2mqtt:info 2019-5-28 23:21:53 Currently 0 devices are joined:
  zigbee2mqtt:warn 2019-5-28 23:21:53 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-5-28 23:21:53 Allowing new devices to join.
  zigbee2mqtt:warn 2019-5-28 23:21:53 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-5-28 23:21:53 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-5-28 23:21:53 Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 2019-5-28 23:21:53 zigbee-shepherd ready

So there must be a problem in the zigbee adapter

bl4ckwhiteeagle, 2019-05-28:
The CC2531 Z-Stack_Home_1.2 CC2531_20190425.zip 	Add sbl firmwares. #90 	3 hours ago
Get the dongle to work:
zigbee-shepherd ready. version: 2.6.3 rev 20190425

bl4ckwhiteeagle, 2019-05-28:
FYI - But this problem is not relevant here in this opened issue
:::
Falling back to old FW zigbee-shepherd ready. version: 2.6.3 rev 20190315
as of timeouts and connection problems / lamps switching on but not off and others 2nd time not on / network not rebuilding fast after ~10 Minutes:

  | Zigbee publish to '0xyz', genOnOff - off - {} - 1 failed  with error Error: AF data request fails, status code: 183. APS no ack.
-- | --
:::

kirovilya, 2019-05-29:
@bl4ckwhiteeagle hi. Did you try dev-branch of iobroker.zigbee adapter? there are the last "shepherd" and "converters".
install it from github https://github.com/ioBroker/ioBroker.zigbee/tarball/dev

bl4ckwhiteeagle, 2019-05-30:
Sadly i get the same error:

ser  stateChange zigbee.0.info.connection  {"val":false,"ack":false,"ts":1559204962433,"q":0,"from":"system.adapter.zigbee.0","user":"system.user.admin","lc":1559204335147}
--
zigbee.0 | 2019-05-30 10:29:22.435 | silly | inMem message  zigbee.0.* zigbee.0.info.connection
zigbee.0 | 2019-05-30 10:29:22.435 | debug | User  stateChange zigbee.0.info.pairingMessage {"val":"Error: Error while  starting zigbee-shepherd!. Error: rsp error:  2","ack":false,"ts":1559204962433,"q":0,"from":"system.adapter.zigbee.0","user":"s
zigbee.0 | 2019-05-30 10:29:22.435 | silly | inMem message  zigbee.0.* zigbee.0.info.pairingMessage
zigbee.0 | 2019-05-30 10:29:22.433 | error | Error while starting zigbee-shepherd!. Error: rsp error: 2
zigbee.0 | 2019-05-30 10:29:17.207 | info | Starting zigbee-shepherd
zigbee.0 | 2019-05-30 10:28:17.201 | info | Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
zigbee.0 | 2019-05-30 10:28:13.212 | debug | system.adapter.admin.0: logging true
zigbee.0 | 2019-05-30 10:28:13.211 | silly | inMem message  *.logging system.adapter.admin.0.logging
zigbee.0 | 2019-05-30 10:28:12.684 | info | Reset coordinator
zigbee.0 | 2019-05-30 10:28:12.672 | info | Queue is: true
zigbee.0 | 2019-05-30 10:28:12.672 | info | Start on port: /dev/ttyACM0 channel 25
zigbee.0 | 2019-05-30 10:28:12.656 | info | starting. Version 0.10.999 in /opt/iobroker/node_modules/iobroker.zigbee, node: v10.15.3

zigbee2mqtt sees the device:
zigbee2mqtt:info 2019-5-30 10:31:39 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-30.10-31-39'
  zigbee2mqtt:info 2019-5-30 10:31:40 Starting zigbee2mqtt version 1.4.0 (commit #927c4db)
  zigbee2mqtt:info 2019-5-30 10:31:40 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-5-30 10:31:55 zigbee-shepherd started
  zigbee2mqtt:info 2019-5-30 10:31:55 Coordinator firmware version: '20190425'
  zigbee2mqtt:info 2019-5-30 10:31:55 Currently 0 devices are joined:
  zigbee2mqtt:warn 2019-5-30 10:31:55 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-5-30 10:31:55 Allowing new devices to join.
  zigbee2mqtt:warn 2019-5-30 10:31:55 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-5-30 10:31:55 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-5-30 10:31:55 Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 2019-5-30 10:31:55 zigbee-shepherd ready

kirovilya, 2019-05-30:
@bl4ckwhiteeagle let's go to issue https://github.com/ioBroker/ioBroker.zigbee/issues/304 
set zigbee-adapter log level to **debug** and then put log file to this issue

bl4ckwhiteeagle, 2019-06-03:
Hi again and thaks for the help so far!

I tried again step by step and now I see a js-controller update to v1.5.12, which didn't show yet after updating the whole /opt/iobroker stuff and the CC2531 got now recognized :)

zigbee-shepherd ready. version: 2.7.2 rev 20190425

Conclusion:
js-controller < 1.5.12 creates this error:
Error while starting zigbee-shepherd!. Error: rsp error: 2

What i've done now?:

cd /opt/iobroker
sudo iobroker stop
sudo iobroker update
sudo iobroker upgrade self
sudo iobroker upload all                  (....for safety, as i'm already on all latest versions)
sudo iobroker start

It works now with my iobroker.dev test-system :)

I'll debug now further and close this issue, if there are no related errors again

bl4ckwhiteeagle, 2019-06-03:
One WXKG02LM - Xiaomi Aqara double key wireless wall switch (EndDevice) with
one LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router) are working fine together. They paired quite as normal with the first try.

The lamp switched with the left button on/off and changed with the right key a random color without any problems.
I'll add more Lamps/Switches and some Osram Plug 01 in the next days

Errors found/solved:
WebGUI Image not found - ../adapter/zigbee/img/zbt_e27_rgbw.png

Solution:
Another iobroker upload all - resolved it

Did many restarts of the iobroker, zigbee-adapter and reconnect the USB CC2531 and the lamp as the switch got pairing and worked after about a minute!

I'll test now by rsyncing to my ESXI VM and see if there is the error gone too

bl4ckwhiteeagle, 2019-06-03:
Copying /opt/iobroker and installing the same Node.js v10.x on ESXI-VM and passthrough the USB CC2531 got the single Lamp and Switch working on productive system out of the box without any reboot.
Dmesg Log looks fine:
[72889.964529] usb 2-2.1: new full-speed USB device number 5 using uhci_hcd
[72890.371576] usb 2-2.1: New USB device found, idVendor=0451, idProduct=16a8
[72890.371578] usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[72890.371580] usb 2-2.1: Product: TI CC2531 USB CDC
[72890.371581] usb 2-2.1: Manufacturer: Texas Instruments
[72890.371582] usb 2-2.1: SerialNumber: __0X00<blanked>
[72890.375543] cdc_acm 2-2.1:1.0: ttyACM0: USB ACM device

So i could upgrade now, if i find some more time next days.

closed 2020-08-18 by github-actions[bot]