Koenkk/Z-Stack-firmware #93: Groups stability

apeeters, 2019-06-10:
I recently updated from 20190315 to 20190425 and zigbee2mqtt 1.4.0 which seems to completely fix crashes (jeej!). Unfortunately the network seems to be a lot less stable: many group commands are not executing. I see "Zigbee publish to group" log messages but nothing happens.

Most of the time no error messages, only every now and then I see "rsp error: 17".

Koenkk, 2019-06-10:
This firmware is known to have some issues.

Could you try the new firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin 

If you have a larger network of +-40 devices I would recommend the source routing one (`20190610`)

apeeters, 2019-06-11:
CC2531_SOURCE_ROUTING_20190610 Seems to work slightly better, but the crashes are back again (request timeout after a few minutes/commands).

apeeters, 2019-06-11:
CC2531_DEFAULT_20190608 Same results.

Koenkk, 2019-06-11:
Can you try with https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190611.zip

apeeters, 2019-06-11:
Seems a tad better, but still crashes with timeout errors.

Koenkk, 2019-06-12:
Can you provide the logs? (`log_level: debug`)

nizr9, 2019-06-16:
I was also seeing both `rsp error: 17` and `rsp error: 16` multiple times per day.  This was on both coordinator firmware versions 20190425 and 20190515.  I was tried the USB stick with a CC2531 and a USB TTL cable for the CC2530 + CC2591 as the coordinator.  With the CC2530 + CC2591 as the coordinator, it crashed about every day (so I moved my coordinator back to a USB CC2531.

Log snippet with the CC2531 source routing firmware (20190611):
```Jun 14 00:24:58 HOST npm[18926]: #033[31m  zigbee2mqtt:error#033[39m 6/14/2019, 12:24:58 AM Zigbee publish to device '0xID', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 183. APS no ack.
Jun 14 00:24:58 HOST npm[18926]: #033[32m  zigbee2mqtt:info#033[39m 6/14/2019, 12:24:58 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF
 data request fails, status code: 183. APS no ack.","meta":{"entity":{"ID":"0xID","type":"device","friendlyName":"test_light"},"message":"{\"state\": \"ON\"}"}}'
Jun 14 00:24:59 HOST kernel: [1295303.320564] usb 3-2: USB disconnect, device number 8
Jun 14 00:24:59 HOST kernel: [1295303.320921] cdc_acm 3-2:1.0: failed to set dtr/rts
Jun 14 00:25:01 HOST npm[18926]: /opt/zigbee2mqtt/node_modules/q/q.js:155
Jun 14 00:25:01 HOST npm[18926]:                 throw e;
Jun 14 00:25:01 HOST npm[18926]:                 ^
Jun 14 00:25:01 HOST npm[18926]: Error: ccznp has not been initialized yet
Jun 14 00:25:01 HOST npm[18926]:     at CcZnp.request (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:183:15)
Jun 14 00:25:01 HOST npm[18926]:     at CcZnp.(anonymous function) [as zdoRequest] (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:225:21)
Jun 14 00:25:01 HOST npm[18926]:     at Zdo._sendZdoRequestViaZnp (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:39:12)
Jun 14 00:25:01 HOST npm[18926]:     at Zdo._genericRequest (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:65:10)
Jun 14 00:25:01 HOST npm[18926]:     at Zdo._specialRequest (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:82:25)
Jun 14 00:25:01 HOST npm[18926]:     at Zdo.request (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:28:21)
Jun 14 00:25:01 HOST npm[18926]:     at Controller.request (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/controller.js:366:19)
Jun 14 00:25:01 HOST npm[18926]:     at req (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/query.js:158:47)
Jun 14 00:25:01 HOST npm[18926]:     at performReq (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/query.js:165:9)
Jun 14 00:25:01 HOST npm[18926]:     at /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/query.js:169:18
Jun 14 00:25:01 HOST npm[18926]: npm ERR! code ELIFECYCLE
Jun 14 00:25:01 HOST npm[18926]: npm ERR! errno 1
Jun 14 00:25:01 HOST npm[18926]: npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
Jun 14 00:25:01 HOST npm[18926]: npm ERR! Exit status 1
Jun 14 00:25:01 HOST npm[18926]: npm ERR!
Jun 14 00:25:01 HOST npm[18926]: npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
Jun 14 00:25:01 HOST npm[18926]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Jun 14 00:25:01 HOST npm[18926]: npm ERR! A complete log of this run can be found in:
Jun 14 00:25:01 HOST npm[18926]: npm ERR!     /home/USER/.npm/_logs/2019-06-14T06_25_01_418Z-debug.log
Jun 14 00:25:01 HOST systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jun 14 00:25:01 HOST systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Jun 14 00:25:01 HOST systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jun 14 00:25:01 HOST systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Jun 14 00:25:01 HOST systemd[1]: Stopped zigbee2mqtt.
Jun 14 00:25:01 HOST systemd[1]: Started zigbee2mqtt.
Jun 14 00:25:01 HOST hass[18808]: 2019-06-14 00:25:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139752049267320] Client exceeded max pending messages [2]: 512
Jun 14 00:25:01 HOST hass[18808]: 2019-06-14 00:25:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139752049267320] Client exceeded max pending messages [2]: 512
Jun 14 00:25:01 HOST hass[18808]: 2019-06-14 00:25:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139752049267320] Client exceeded max pending messages [2]: 512
Jun 14 00:25:01 HOST hass[18808]: 2019-06-14 00:25:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139752049267320] Client exceeded max pending messages [2]: 512
Jun 14 00:25:01 HOST hass[18808]: 2019-06-14 00:25:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139752049267320] Client exceeded max pending messages [2]: 512
Jun 14 00:25:01 HOST hass[18808]: 2019-06-14 00:25:01 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.139752049267320] Client exceeded max pending messages [2]: 512
```

Node debug log 
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.4.0~prestart: zigbee2mqtt@1.4.0
6 info lifecycle zigbee2mqtt@1.4.0~start: zigbee2mqtt@1.4.0
[... cut ...] vice (even single bulbs) into zigbee2mqtt groups and then setting up the automation in home assistant to be an MQTT trigger to the specific topic and a condition of type template `{{ trigger.payload_json.click == "single" }}` with a call to light.toggle to the zigbee2mqtt group id for entity_id.  Now the 'click' just sends a single toggle, instead of one for 'click' and the other for '' (null), then for home assistant to add another for each.  Another bonus is that it is much faster now as well.

apeeters, 2019-06-18:
@Koenkk Please find the crashing part of the log below, when using CC2531_SOURCE_ROUTING_20190611. As @nzr97 also noticed, this happens mostly when sending several commands in a row (in this case turning on 6 zigbee groups at once). This did not happen on 20190425.

```
  zigbee2mqtt:debug 6/18/2019, 8:54:10 PM Received MQTT message on 'zigbee2mqtt/eetkamer_staanlamp/set' with data '{"state": "OFF"}'
  zigbee2mqtt:info 6/18/2019, 8:54:10 PM Zigbee publish to group '12', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 6/18/2019, 8:54:10 PM MQTT publish: topic 'zigbee2mqtt/eetkamer_staanlamp', payload '{"state":"OFF","brightness":255,"color_temp":252}'
  zigbee2mqtt:debug 6/18/2019, 8:54:19 PM Received MQTT message on 'zigbee2mqtt/eetkamer_staanlamp/set' with data '{"state": "ON"}'
  zigbee2mqtt:debug 6/18/2019, 8:54:19 PM Received MQTT message on 'zigbee2mqtt/eetkamer_staanlamp/set' with data '{"state": "ON", "brightness": 255, "color_temp": 252}'
  zigbee2mqtt:info 6/18/2019, 8:54:19 PM Zigbee publish to group '12', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 6/18/2019, 8:54:19 PM MQTT publish: topic 'zigbee2mqtt/eetkamer_staanlamp', payload '{"state":"ON","brightness":255,"color_temp":252}'
  zigbee2mqtt:info 6/18/2019, 8:54:20 PM Zigbee publish to group '12', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 6/18/2019, 8:54:20 PM MQTT publish: topic 'zigbee2mqtt/eetkamer_staanlamp', payload '{"state":"ON","brightness":255,"color_temp":252}'
  zigbee2mqtt:info 6/18/2019, 8:54:20 PM Zigbee publish to group '12', lightingColorCtrl - moveToColorTemp - {"colortemp":252,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 6/18/2019, 8:54:20 PM MQTT publish: topic 'zigbee2mqtt/eetkamer_staanlamp', payload '{"state":"ON","brightness":255,"color_temp":252}'
  zigbee2mqtt:debug 6/18/2019, 8:54:21 PM Received MQTT message on 'zigbee2mqtt/eetkamer_groot/set' with data '{"state": "ON"}'
  zigbee2mqtt:debug 6/18/2019, 8:54:21 PM Received MQTT message on 'zigbee2mqtt/eetkamer_groot/set' with data '{"state": "ON", "brightness": 255, "color_temp": 252}'
  zigbee2mqtt:info 6/18/2019, 8:54:21 PM Zigbee publish to group '5', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 6/18/2019, 8:54:21 PM MQTT publish: topic 'zigbee2mqtt/eetkamer_groot', payload '{"state":"ON","brightness":204,"color_temp":406}'
  zigbee2mqtt:info 6/18/2019, 8:54:22 PM Zigbee publish to group '5', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 6/18/2019, 8:54:22 PM MQTT publish: topic 'zigbee2mqtt/eetkamer_groot', payload '{"state":"ON","brightness":255,"color_temp":406}'
  zigbee2mqtt:debug 6/18/2019, 8:54:22 PM Received zigbee message of type 'attReport' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'TRADFRI bulb GU10 WS 400lm' (0x90fd9ffffe00f039) of endpoint 1
  zigbee2mqtt:info 6/18/2019, 8:54:22 PM MQTT publish: topic 'zigbee2mqtt/eetkamer_f', payload '{"state":"ON","brightness":254,"color_temp":375,"linkquality":2,"color_mode":2,"color":{"x":0.46,"y":0.411}}'
  zigbee2mqtt:debug 6/18/2019, 8:54:22 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'TRADFRI bulb GU10 WS 400lm' (0x90fd9ffffe00f039) of endpoint 1
  zigbee2mqtt:info 6/18/2019, 8:54:22 PM Zigbee publish to group '5', lightingColorCtrl - moveToColorTemp - {"colortemp":252,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:debug 6/18/2019, 8:54:22 PM Received zigbee message of type 'attReport' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'TRADFRI bulb GU10 WS 400lm' (0x90fd9ffffe8ea715) of endpoint 1
  zigbee2mqtt:info 6/18/2019, 8:54:22 PM MQTT publish: topic 'zigbee2mqtt/eetkamer_c', payload '{"state":"ON","brightness":254,"color_temp":375,"linkquality":13,"color":{"x":0.46,"y":0.411},"color_mode":2}'
  zigbee2mqtt:debug 6/18/2019, 8:54:22 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'TRADFRI bulb GU10 WS 400lm' (0x90fd9ffffe8ea715) of endpoint 1
  zigbee2mqtt:info 6/18/2019, 8:54:22 PM MQTT publish: topic 'zigbee2mqtt/eetkamer_groot', payload '{"state":"ON","brightness":255,"color_temp":252}'
  zigbee2mqtt:debug 6/18/2019, 8:54:34 PM Received MQTT message on 'zigbee2mqtt/gangetje/set' with data '{"state": "ON"}'
  zigbee2mqtt:info 6/18/2019, 8:54:34 PM Zigbee publish to group '3', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:debug 6/18/2019, 8:54:34 PM Received MQTT message on 'zigbee2mqtt/living_groot/set' with data '{"state": "ON"}'
  zigbee2mqtt:debug 6/18/2019, 8:54:34 PM Received MQTT message on 'zigbee2mqtt/kast/set' with data '{"state": "ON"}'
  zigbee2mqtt:debug 6/18/2019, 8:54:35 PM Received MQTT message on 'zigbee2mqtt/tv/set' with data '{"state": "ON"}'
  zigbee2mqtt:debug 6/18/2019, 8:54:35 PM Received MQTT message on 'zigbee2mqtt/zetel/set' with data '{"state": "ON"}'
  zigbee2mqtt:info 6/18/2019, 8:54:35 PM Zigbee publish to group '4', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
[... cut ...]
  zigbee2mqtt:info 6/18/2019, 8:54:47 PM Zigbee publish to group '7', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 6/18/2019, 8:54:49 PM Zigbee publish to group '8', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
  zigbee2mqtt:info 6/18/2019, 8:54:49 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: request timeout","meta":{"entity":{"ID":8,"type":"group","friendlyName":"zetel"},"message":"{\"state\": \"ON\"}"}}'
```

Koenkk, 2019-06-19:
@apeeters although you didn't have the errors with `20190425`, you also had group errors with it (OP). For debugging purposes, could you try increasing https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/util/zigbeeQueue.js#L2 to e.g. `const delay = 500;` and see if it still happens?

apeeters, 2019-07-01:
@Koenkk unfortunately I cannot easily change the delay setting on hassio. If I missed something, just let me know.

I did upgrade to `SOURCE_ROUTING_20190619` today and it was a lot more stable than `20190611` but eventually failed as well with timeouts.

Koenkk, 2019-07-02:
I've added the queue settings, I would propose to test with the following:

**1**
```js
queue:
  delay: 250
  simultaneously: 3
```

**2**
```js
queue:
  delay: 500
  simultaneously: 5
```

**3**
```js
queue:
  delay: 500
  simultaneously: 3
```

Please let me know if any of the configurations fixes your problem.

apeeters, 2019-07-11:
@Koenkk I ran all three options for over 24 hours without a single crash. Now back to default queue settings and still no crashes. Whichever setting I select, messages do not seem to arrive properly at all Tradfri lights. I use group commands and lights react one after another instead of simultaneously. Some lights even fail to apply the change.

This is all on CC2531_SOURCE_ROUTING_20190619 with zigbee2mqtt-edge addon of approx 2nd of July.

Koenkk, 2019-07-12:
Zigbee2mqtt does not have control over the timing, it's a zigbee broadcast message which each devices applies to itself if it's in the group (see http://www.zigbee2mqtt.io/information/groups.html#how-do-groups-work). 

Having delays or failures could indicate that your network has a lot of interference or range issues (http://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range.html)

apeeters, 2019-07-12:
All lights that should switch simultaneously are really close together, so it seems to be more of a suboptimal routing issue than a range issue.

Could it be caused by the lights reporting their state and flooding the network? I will add a comment about this in https://github.com/Koenkk/zigbee2mqtt/issues/1064.

Koenkk, 2019-07-12:
That may be the cause

apeeters, 2019-07-30:
Disabling reporting by powercycling the Tradfri devices fixed 99% of my issues, excellent!

closed 2020-08-18 by github-actions[bot]