Koenkk/Z-Stack-firmware #96: Question about cmpling the firmware

LengedZhao, 2019-06-20:
I set it up according to COMPILE.md to cmplie the FIRMWARE_CC2530 myself, but there are two question I have meet.
First, I set Stack sizes XDATA as 0x400, but IAR reports "Error[e16]: Segment XDATA_I (size: 0xc1 align: 0) is too long for segment definition. At least 0x5b more bytes needed. ...". So I set XDATA as 0x390, it cmplied sueccessly.
Second, PC can recive two data sets(FE 06 41 80 01 02 00 02 07 00 C1 )after restart CC2530. Then, I send some commands to CC2530 by serial ports, but CC2530 responses nothing except of command SYS_RESET_REQ.
This is everything ok when I use your CC2530ZNP-with-SBL.hex. Are there any settings that have been ignored?

Koenkk, 2019-06-21:
What IAR version are you using?

LengedZhao, 2019-06-24:
I have tried IAR 10.20.1 with Z-Stack3.0.2 and IAR 9.30.1 with Z-Stack 3.0.0 and Z-Stack homg 1.2.2a

closed 2020-08-18 by github-actions[bot]